ros-perception/vision_opencv #316: '/usr/include/opencv'  not found!

JeyP4, 2020-02-14:
Hi
My system is Jetson Nano ubuntu 18.04. Ros-melodic.
Jetson came with opencv 4.1.1. And during ROS installation, it did not install opencv3.
But Seperately, I source build opencv3.4.0. So now I have two opencvs.

[image]

To solve above problem, I tried to modify `/opt/ros/melodic/lib/pkgconfig/cv_bridge.pc`. Here I edited include and libs pointing opencv3.4.0.
How can I recompile cv_bridge to make the changes effective.

My `/usr/include` looks like this:
[image]

JeyP4, 2020-02-14:
sudo apt purge ros-melodic-cv-bridge
cd catkin_ws/
git clone https://github.com/ros-perception/vision_opencv.git
# ensure `pkg-config --modversion opencv` able to find OPENCV3.4.0
catkin_make -DCMAKE_BUILD_TYPE=Release

closed 2020-02-14 by JeyP4

henryfhq-drone, 2020-05-14:
@JeyP4  Hi

I also encounter the same issue as you and I tried to run the solution as mentioned above. However, the issue still persist. Here is the picture:

[image]

So is there a way that I can resolve it?
I am using the Jetson Nano and I'm using Ubuntu 18.04 and ROS version is melodic.

Thank you very much.

phoenixrider12, 2022-06-03:
Hey @JeyP4 
how did you install opencv 3.4?