ros-perception/vision_opencv #318: Libboost_python error: undefined reference on branch ros2

jdcdp, 2020-02-19:
I am getting the following error when building vision_opencv on ros2 dashing (debian 10):
```                                                           
--- stderr: cv_bridge                             
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyUnicode_InternFromString'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyLong_FromLong'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyModule_Create2'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_InPlaceRemainder'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PySlice_New'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyList_Sort'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyObject_SetAttrString'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyType_IsSubtype'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyList_Insert'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyExc_RuntimeError'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyComplex_Type'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyDict_Copy'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyEval_CallFunction'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyDict_Size'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyUnicode_Type'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyStaticMethod_New'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyList_Reverse'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyErr_NoMemory'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `_Py_fopen'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyUnicode_FromString'
...(etc)
make[2]: *** [test/CMakeFiles/cv_bridge-utest.dir/build.make:184: test/cv_bridge-utest] Error 1
make[1]: *** [CMakeFiles/Makefile2:239: test/CMakeFiles/cv_bridge-utest.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< cv_bridge	[ Exited with code 2 ]

```

This is my libboost linking:
```-rw-r--r--   1 root root   650044 Jan 26 20:20 libboost_python27.a
lrwxrwxrwx   1 root root       27 Jan 26 20:20 libboost_python27.so -> libboost_python27.so.1.67.0
-rw-r--r--   1 root root   284168 Jan 26 20:20 libboost_python27.so.1.67.0
-rw-r--r--   1 root root   642638 Jan 26 20:20 libboost_python37.a
lrwxrwxrwx   1 root root       27 Jan 26 20:20 libboost_python37.so -> libboost_python37.so.1.67.0
-rw-r--r--   1 root root   280072 Jan 26 20:20 libboost_python37.so.1.67.0
lrwxrwxrwx   1 root root       19 Jan 26 20:20 libboost_python3.a -> libboost_python37.a
lrwxrwxrwx   1 root root       19 Jan 26 20:20 libboost_python3-py37.a -> libboost_python37.a
lrwxrwxrwx   1 root root       20 Jan 26 20:20 libboost_python3-py37.so -> libboost_python37.so
lrwxrwxrwx   1 root root       20 Jan 26 20:20 libboost_python3.so -> libboost_python37.so
lrwxrwxrwx   1 root root       19 Jan 26 20:20 libboost_python.a -> libboost_python27.a
lrwxrwxrwx   1 root root       20 Jan 26 20:20 libboost_python.so -> libboost_python27.so
```

I also tried using python2.7 and building libboost from source (1.58 and 1.72) with similar results.
This error does not seem to happen on other builds that require libboost

lukicdarkoo, 2020-02-25:
I tried to compile it on Raspbian Buster with Boost 1.67 and Python 3.7 and got the same error. Therefore, I believe the error is related to the packages available on Buster (Debian 10).

Anyhow, I managed to solve it by adding `python3.7m` to the following line:
https://github.com/ros-perception/vision_opencv/blob/c2de79c94ce7bdd3ac4081a6b0f6d4902a342a7b/cv_bridge/src/CMakeLists.txt#L13
so the result should look like:
```cmake
target_link_libraries(${PROJECT_NAME} python3.7m ${Boost_LIBRARIES} ${OpenCV_LIBRARIES})
```

Of course, this is not a proper solution. Further investigation should be made and a more general solution should be found (one that doesn't break builds with Python <3.7).

jdcdp, 2020-02-25:
Thank you. How about this?
```
target_link_libraries(${PROJECT_NAME} ${PYTHON_LIBRARIES} ${Boost_LIBRARIES} ${OpenCV_LIBRARIES})>
```

It seems to work for me, but I have not tested it in other python versions.

 I do no understand why is this a buster exclusive problem though. Is python supposed to link automatically when using boost , is it linked somewhere else or is python not linked at all?

lukicdarkoo, 2020-02-25:
I am not sure, I have been wondering why it wasn't linked explicitly for other platforms as well. On my Ubuntu 18.04 works just fine without any modification.

P.S. Something like this should also work:
```
colcon build --packages-select cv_bridge --cmake-args "-DCMAKE_CXX_STANDARD_LIBRARIES=-lpython3.7m"
```

goekce, 2020-07-27:
I'm on Archlinux and got the same `ldd` errors.

using @lukicdarkoo's advice fixed the error:
```
target_link_libraries(${PROJECT_NAME} python3 ${Boost_LIBRARIES} ${OpenCV_LIBRARIES})
```
but using
```
colcon build --packages-select cv_bridge --cmake-args "-DCMAKE_CXX_STANDARD_LIBRARIES=-lpython"
```
did not help. In this case the compilation just fails without any error message from `ldd`

AchmadFathoni, 2020-08-26:
Any update? My God using ROS2 on tier 3 platform makes me insane.

AchmadFathoni, 2020-08-26:
Well I finally success compiling after adding `${Boost_LIBRARIES}` to my `target_link_libraries` then delete the **install** and **build** folder in main directory.

goekce, 2020-10-07:
I was getting the errors in my above posting while building `gazebo-ros-pkgs`. Today I recompiled them and do not get these errors on Archlinux anymore.

jdcdp, 2020-10-07:
I can't replicate this error anymore, so I guess someone somewhere else fixed it. Closing the issue

jdcdp, 2020-11-29:
This problem still exists according to @akshitj1 . I also had to re-apply the fix proposed here a few days ago.

ijnek, 2022-09-12:
Closing, as the platform stated in the original post has reached EOL. Please comment here to get this re-opened.

closed 2022-09-12 by ijnek