ros-perception/vision_opencv #320: peoblem with instalation

WojciechowskiMarek, 2020-02-25:
CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
  Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.

Where could be error ?

dvigne, 2020-03-04:
Do you have OpenCV installed?

ntomer, 2020-04-09:
I found a similar problem. I Installed opencv with 'sudo apt install python3-opencv python-opencv'.  
`hrrnxt@RUA-X-Ubuntu:~$ sudo apt install python3-opencv python-opencv
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-opencv is already the newest version (3.2.0+dfsg-4ubuntu0.1).
python3-opencv is already the newest version (3.2.0+dfsg-4ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 56 not upgraded.
`
However there is no path '/usr/include/opencv'  but there is '/usr/include/opencv4' and '/usr/include/opencv2'

Should I make a link from '/usr/include/opencv' to '/usr/include/opencv4'?

Benefe, 2020-04-15:
I have the same problem. Like @ntomer there is also only '/usr/include/opencv4/opencv2' instead of '/usr/include/opencv'.
I am using a Jetson TX2 with Jetpack 4.2 and ROS Melodic.
I copied the branch of #321 and edited the path for OpenCV in `cv_BridgeConfig.cmake` in line 94 and 96.
But now I get an Error saying there is the file `cv.h` missing.
I keep on digging, but I hope someone already managed it.

Benefe, 2020-04-15:
I've found a potential solution mentioned in #209 and #316, but `cv.h` still couldn't be found.

ntomer, 2020-04-15:
I think there has been a change of opencv version behind the scenes.  

Once I updated the path I had to modify the package that was using 'vision_opencv' with:
(remove)#include <cv.h>
(add)#include "opencv2/opencv.hpp"

And then remove some code which looks like it has been outdated.  like the following lines:
namedWindow("Acquisition", CV_WINDOW_NORMAL | CV_WINDOW_KEEPRATIO);
cvDestroyAllWindows();

This all compiled before for me.  

When I run:
python2 -c "import cv2; print(cv2.__version__)"
and
python3 -c "import cv2; print(cv2.__version__)"
the output is 4.0.0
but listing the python installs with:  apt list python*opencv
Listing... Done
python-opencv/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed]
python3-opencv/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed]

suggesting the package is called 3.2 but programmatically identified as 4.0?

That would explain why #321 fixes the problem.

AlexW159, 2020-04-17:
Hey. I have the same problem.
But sorry, I do not really get how you both solved it... 

Clone the 'vision_opencv' packet from @OAkyildiz into `~/catkin_ws/src` ....
After `catkin build` I get the next CMake Error: 
`<pre><font color="#EF2929"><b>CMake Error</b></font> at /home/nuc-falke/catkin_ws/src/vision_opencv/cv_bridge/CMakeLists.txt:16 (find_package):
  Could not find a configuration file for package &quot;OpenCV&quot; that is compatible
  with requested version &quot;3&quot;.

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/opencv4/OpenCVConfig.cmake, version: 4.1.1
</pre>`

What am I doing wrong?
Or is there something else faulty?

Benefe, 2020-04-17:
You might need to edit the required OpenCV version from 3 to 4 in the CMakeLists.

AlexW159, 2020-04-18:
Is there a significant difference between `catkin build` and `catkin_make`? 
As I read, `catkin build` is just a newer and more advanced version of `catkin_make` (and some other variations)...

I used `catkin build`, because `catkin_make` has a problem with my non-homogeneous workspace.
I have a mavlink 'package' in it which seems to be `(plain cmake)` (a communication protocol for UAS from PX4). 
And `catkin build` handles that issue; that's why I used it so far. The Error message suggest to use `catkin_make_isolated` which I am not famliar with yet, and I don't know how to use...

I installed `sudo apt purge ros-melodic-cv-bridge`, 
and I changed the `OpenCV 3 REQUIRED COMPONENTS` to `OpenCV 4 REQUIRED COMPONENTS`.
Now I am getting cv::MatAllocator errors:
- abstract type 'NumpyAllocator'
- no member named 'refcount'
and so on...
[image]

Benefe, 2020-04-20:
I understand. The file in which the error occurs is `module_opencv2.cpp`, by using opencv2 functions which are not implemented in opencv4.
But as in the `CMakeLists.txt` of cvBridge defined it should use the corresponding file to the version of opencv. In our case opencv4. So the problem is either CMake does not recognize the version of opencv correctly or opencv4 is not installed correct.
I would suggest to clone the vision_opencv repo from @OAkyildiz into an empty directory, so git isn't able to merge files and make sure that only opencv4 is installed or CMake could only access opncv4.

AlexW159, 2020-04-21:
I am kind of confused with my installed opencv versions...
When executing the commands @ntomer mentioned, i get:
`$python2 -c "import cv2; print(cv2.__version__)"`
`4.3.0-dev`
and 
`$python3 -c "import cv2; print(cv2.__version__)"`
`4.2.0`

And like @ntomer said, when requestin the apt for the installed packages, it lists:
`$apt list python*opencv`
`Listing... Done`
`python-opencv/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed]`
`python3-opencv/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed]`

But before I changed the "REQUIRED OpenCV package from 3 to 4 on the CMakeLists.txt, the CMake Error said:

> <pre><font color="#EF2929"><b>CMake Error</b></font> at vision_opencv/cv_bridge/CMakeLists.txt:16 (find_package):
>   Could not find a configuration file for package &quot;OpenCV&quot; that is compatible
>   with requested version &quot;3&quot;.
> 
>   The following configuration files were considered but not accepted:
> 
>     /usr/local/lib/cmake/opencv4/OpenCVConfig.cmake, version: 4.3.0
>     /usr/lib/x86_64-linux-gnu/cmake/opencv4/OpenCVConfig.cmake, version: 4.1.1 </pre>

So it found version 4.3.0 and 4.1.1.

However, with the command I got from [#316 ](https://github.com/ros-perception/vision_opencv/issues/316):
`$ pkg-config --modversion opencv`
`Package opencv was not found in the pkg-config search path.
Perhaps you should add the directory containing 'opencv.pc'
to the PKG_CONFIG_PATH environment variable`
`No package 'opencv' found`
...he says he could not find an opencv package

@Benefe:
I cloned the vision_opencv repo from @OAkyildiz into an empty directory and empty catkin_ws,
so now I am also able to run catkin_make. However, the errors remain the same.
Regarding your last comment: How can I check which version CMake recognizes? Or might this also be caused by the suggestion @ntomer made : 
> package is called 3.2 but programmatically identified as 4.0?

Or how do I remove/deinstall all opencv files from my system, so that I can do a plain reinstall?

Benefe, 2020-04-23:
In fact it is quite confusing with your installed opencv versions and with the versions make can find.
`sudo find / -name "*opencv*" -exec rm -i {} \;` 
as in [Removing all installed OpenCV libs
](https://stackoverflow.com/questions/9276169/removing-all-installed-opencv-libs) recommended should do the job.

I think with your command:
> `$apt list python*opencv`
> `Listing... Done`
> `python-opencv/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed]`
> `python3-opencv/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 amd64 [installed]`
> 
you get the python wrapped opencv libraries.

espentrydal, 2020-04-23:
I fixed it with editing /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake and then replacing opencv with opencv4 on line 96 so it becomes set(_include_dirs "include;/usr/include;/usr/include/opencv4").

Benefe, 2020-04-23:
Nice! Thanks, I totally forgot about that.

AlexW159, 2020-04-24:
@espentrydal, thank you, that fix sounds reasonable as well.

 However, I actually decided to remove OpenCV completely and go back to version 3.4 for now, which works perfectly fine. But I will keep that in mind.

Thanks to all from my side.

OAkyildiz, 2020-05-09:
> In fact it is quite confusing with your installed opencv versions and with the versions make can find.
> `sudo find / -name "*opencv*" -exec rm -i {} \;`
> as in [Removing all installed OpenCV libs [...]

> I think there has been a change of opencv version behind the scenes.
>
> Once I updated the path I had to modify the package that was using 'vision_opencv' with:
> (remove)#include <cv.h>
> (add)#include [...]

I am looking into it right now, but what you are seeing might be the OpenCV that comes with ROS being linked

OAkyildiz, 2020-05-09:
I have installed OpenCV 4.1 again when I flashed my JetsonTX2 with Nvidia SDK manager. After installing ROS I was able to `catkin build cv_bridge` with my fork for #321.  

If you are trying to build other packages while having issues with this you can also use '--start-with' argument but it is more likely to be a CMakeFile error

OAkyildiz, 2020-11-21:
UPDATE: I received the same error. Removing built packages and rebuilding workspace  solved it:
`catkin clean -bldy #use [catkin_tools]( https://catkin-tools.readthedocs.io/en/latest/installing.html)`

`catkin build -j`