ros-perception/vision_opencv #329: Problem with the Cmake issue Involving cv_bridge

henryfhq-drone, 2020-05-13:
Hi @vrabaud 
I have Installed the cv_bridge thing to my catkin_ws and I tried to do catkin_make for a github code that I have to download as part of my project.

However, I ran into a cmake issue that says that the usr/include/opencv is not found. The error is in the picture below:

[image]

I have tried download the opencv file but it is to no avail. and I have also tried to change a cmake file but I cannot edit.

I don't know is there an issue on the cv_bridge or is there something else that is missing?

I'm running on Jetson Nano which comes with opencv version 4.1.1. OS version is Ubuntu 18.04 and my ROS version is melodic. 

Thank you very much.

debackerj93, 2020-05-21:
I've had the same problem.. anyone have an idea of what's going on?

henryfhq-drone, 2020-05-22:
@isokar @debackerj93 I would like to know if you guys using the same specification as me? If yes, then the following information might help you guys with solving the problem.

Jetson Nano have change the file name from opencv to opencv4. However, in the cv_config.cmake file, at line 100+ to 113 (forgot which line). The code still says to include opencv file. There is also no way to change it since it is a root file on the Jetson Nano. 

For my case, 
Firstly, I source build opencv 3.4.0. (If cannot download, I change the errant line to include Eigen 3). Next, I download the vision opencv (this one), extract it to the home directory and then 
I clone the cv_bridge to the source file of the catkin workspace then catkin make. 

It is important to know that you need to install the following by running the following codes in the terminal:
sudo apt-get install ros-<Your ROS version> - vision-opencv (for my case, melodic)
sudo apt-get install ros - <Your ROS version> - cv-bridge. (for my case, melodic)

Thanks, everyone who has indicated that you shared the same problem as me. I really hope the above information helps. I know that it is very frustrating as well because you cannot implement the new codes.

cafgiu, 2020-05-22:
Hi @henryfhq-drone. I had the same problem. I have installed ros-melodic-vision-opencv and ros-melodic-cv-bridge with apt. I'm running on Jetson Nano. 
What do you mean by 
> Firstly, I source build opencv 3.4.0. (If cannot download, I change the errant line to include Eigen 3).

Can you explain me?
Sorry, but I'm a newbie.

Thank you.

henryfhq-drone, 2020-05-22:
Hi @cafgiu 

Thanks for your message. Regarding the statement on source building the opencv 3.4.0. What I mean is you download the opencv using the link here:
 
https://github.com/kyamagu/mexopencv/wiki/Installation-%28Linux%2C-MATLAB%2C-OpenCV-3%29#1-opencv 

However, if you encounter errors, with the code as shown below:

[image]

Try to change the errant line to include < eigen3/ > at the front after the #include. 

Don't worry, I am also a newbie to it as well.

briangmoeller, 2020-05-22:
Hi

I just made an symbolic link to the missing folder.

```
cd /usr/include/
sudo ln -s /usr/include/opencv4 opencv
```

And it works for me

cafgiu, 2020-05-25:
Hi @henryfhq-drone 

Thanks for your answer. I have followed your steps but  I have the below error 

[image]

Can you help me? 
Thank you.

henryfhq-drone, 2020-05-25:
@cafgiu Do you have python 3 installed? (version at least 3.6 if I am not wrong) 

If yes then that seems weird

cafgiu, 2020-05-25:
The command apt list --installed python* show me the below packages installed

https://pastebin.com/M1yp1Fj6

BenArtes, 2020-05-28:
I had this same issue. I tracked it down to using Nvidia SDKManager to Flash an Nvidia Jetson / Xavier. I was trying to use Jetpack 4.3 or 4.4, and in the process it replaced OpenCV 2 with OpenCV 4; this caused the cv-bridge not found issue.

Not knowing what had installed opencv4, I removed all ros-packages / opencv and reinstalled and was able to build but I was left without CUDA; upon using SDKManager to reinstall the cuda toolchain I ran into the same issue.

I found that if I installed Jetpack 4.2 using SDKManager it would remove opencv4 and resolve the issue.

cafgiu, 2020-05-28:
I have resolved commenting the row 11 12 in the CMakeLists.txt in the package cv_bridge 

```
if(PYTHONLIBS_VERSION_STRING VERSION_LESS "3.8")
    # Debian Buster
    #find_package(Boost REQUIRED python37)
    #else()
    # Ubuntu Focal
    find_package(Boost REQUIRED python)
  endif()
```

and adding 0 at the #define NUMPY_IMPORT_ARRAY_RETVAL in the file /usr/include/python2.7/numpy/__multiarray_api.h

[image]

jeffav, 2020-05-30:
Same issue here on a Xavier NX with Jetpack 4.4 DP. I was able to resolve it by downgrading opencv-dev.
Find what versions are available to you
```
$ apt list -a libopencv-dev
Listing... Done
libopencv-dev/stable 4.1.1-2-gd5a58aa75 arm64 [upgradable from: 3.2.0+dfsg-4ubuntu0.1]
libopencv-dev/bionic-updates,bionic-security,now 3.2.0+dfsg-4ubuntu0.1 arm64 [installed,upgradable to: 4.1.1-2-gd5a58aa75]
libopencv-dev/bionic 3.2.0+dfsg-4build2 arm64
```
Install the latest one before 4.0. For me it was:
`$ sudo apt install libopencv-dev=3.2.0+dfsg-4ubuntu0.1
`

tamisalex, 2020-06-01:
I solved this by installing libopencv as @jeffav described and then with the changes that @cafgiu describes so.

1. Downgrade to opencv 3
2. Changed CMakeLists.txt  and __multiarray_api.h as in cafgiu post

jebasamuel, 2020-06-04:
For opencv issue, i tried different method. It resolves the errors.

$sudo ln -s /usr/include/opencv4/ /usr/include/opencv

cafgiu, 2020-06-04:
Hi @jebasamuel. I tried this method but it not resolve the errors for me. 

> 
> 
> For opencv issue, i tried different method. It resolves the errors.
> 
> $sudo ln -s /usr/include/opencv4/ /usr/include/opencv

jebasamuel, 2020-06-04:
@cafgiu Do you still have cv_bridge.cmake error saying unable to find opencv? Could you please post the catkin build log?

IhsanFikri, 2020-06-30:
> Hi @henryfhq-drone
> 
> Thanks for your answer. I have followed your steps but I have the below error
> 
> [image]
> 
> Can you help me?
> Thank you.

Hello, i have the same problem like you.

i solve that issue with change the boost requirement with python major

find_package(Boost REQUIRED python3)

stevenlee090, 2020-07-15:
> For opencv issue, i tried different method. It resolves the errors.
> 
> $sudo ln -s /usr/include/opencv4/ /usr/include/opencv

Can confirm that this works.

GilenTell, 2020-07-16:
> > For opencv issue, i tried different method. It resolves the errors.
> > $sudo ln -s /usr/include/opencv4/ /usr/include/opencv
> 
> Can confirm that this works.

I can confirm this too with Nvidia Jetson TX2

siddharthcb, 2021-01-31:
i am facing exact issues as @henryfhq-drone and @cafgiu .
can someone please describe the solution for jetson nano 18.04(melodic)

henryfhq-drone, 2021-02-01:
@siddharthcb Try check your Jetson Nano version and then check the version of the cv bridge thing. Because I use an older version of jetson Nano software and then I use melodic for the cv bridge module thing.... 
Oh yeah you have to change one line of a code to a full number (forgot which document and which line)

henryfhq-drone, 2021-02-01:
> @isokar @debackerj93 I would like to know if you guys using the same specification as me? If yes, then the following information might help you guys with solving the problem.
> 
> Jetson Nano has changed the file name from opencv to opencv4. However, in the cv_bridgeConfig.cmake file, at line 100+ to 113 (forgot which line). The code still says to include opencv file. There is also no way to change it since it is a root file on the Jetson Nano.
> 
> For my case,
> Firstly, I source build opencv 3.4.0. (If cannot download, I change the errant line to include Eigen 3). Next, I download the vision opencv (this one), extract it to the home directory and then
> I clone the cv_bridge to the source file of the catkin workspace then catkin make.
> 
> It is important to know that you need to install the following by running the following codes in the terminal:
> sudo apt-get install ros-< Your ROS version > - vision-opencv (for my case, melodic)
> sudo apt-get install ros - < Your ROS version > - cv-bridge. (for my case, melodic)
> 
> Thanks, everyone who has indicated that you shared the same problem as me. I really hope the above information helps. I know that it is very frustrating as well because you cannot implement the new codes.

Solution I have written here for reference @siddharthcb

freesolo92, 2021-10-13:
Hello,

I had similar issues described above. With compiling cv_bridge with OpenCV Version 4.5.1 on a Jetson Nano (Ubuntu 18 and Melodic Distro). I just installed ROS base and not the fully distro.

First error while building was with libboos library for python3
Second error was about empty return value in: `In function ‘void* do_numpy_import()’:
/usr/include/python2.7/numpy/__multiarray_api.h:1537:144: error: return-statement with no value, in function returning ‘void*’ [-fpermissive]`

The way I solved this issues:

- Create sympolic link for opencv: `ln -s /usr/include/opencv4 /usr/include/opencv`
- Download vision_opencv noetic branche: `git clone https://github.com/ros-perception/vision_opencv.git `
- Install libboost with package manager (If you have not already done this): `sudo apt-get install libboost-dev libboost-all-dev`
- Comment line 10 till 13 https://github.com/ros-perception/vision_opencv/issues/329#issuecomment-635192230
- Add number 0 in `/usr/include/python2.7/numpy/__multiarray_api.h` like @cafgiu mentioned above: https://github.com/ros-perception/vision_opencv/issues/329#issuecomment-635192230
- Create a build directory in cv_bridge: `mkdir build && cd build`
- `cmake ..`
- `make -j$(nproc)` (for the Jetson Nano I used all 4 cores)
- Optional: `make install` to install the binaries in the include directory of the cv_bridge package

I hope this may be helpfully for anyone.