ros-perception/vision_opencv #334: [request] Support for OpenCV 4 in ubuntu 18.04

MatthijsBurgh, 2020-05-27:
Could you add support for OpenCV 4 in ubuntu 18.04.

Although it isn't distributed by default, I want to start using OpenCV 4, by installing from source. It would be nice if cv_bridge, etc. would support it.

There are more poeple who want to use it on 18.04, see #327

Algabri, 2020-06-02:
Did you build it with Ubuntu 18.04?  I'd like to build it with 16.04.

MatthijsBurgh, 2020-06-02:
No, I didn't.

PrabhjotKaurGosal, 2020-08-08:
I have tried installing OpenCV 4 on ubnuntu 18.04 fromt source. I can see it was successfully installed (that is , I got no errors when I opened the python2.7 shell and inquired for cv version. I got the message back '4.0.0' , which is what I had installled.  . .  I also have cv_bridge installed for ROS melodic on my machine. Despite this, I am unable to build the package darknet_ros which uses opencv. Any help would be appreciated.

closed 2021-09-21 by MatthijsBurgh