ros-perception/vision_opencv #339: [Foxy] cv_bridge broken

SteveMacenski, 2020-06-12:
Hi,

I noticed in `image_pipeline`'s CI that there's an unreported exception due to `cv_bridge`.

```
12:17:57     from camera_calibration.calibrator import MonoCalibrator, StereoCalibrator, \
12:17:57 src/camera_calibration/calibrator.py:41: in <module>
12:17:57     import cv_bridge
12:17:57 /opt/ros/foxy/lib/python3.8/site-packages/cv_bridge/__init__.py:6: in <module>
12:17:57     from cv_bridge.boost.cv_bridge_boost import cvtColorForDisplay, getCvType
12:17:57 E   SystemError: initialization of cv_bridge_boost raised unreported exception
12:17:57 ---- generated xml file: /tmp/ws/test_results/camera_calibration/pytest.xml ----
```

I suspected CI issues, but in an ipython3 session locally I see the same thing with released debians

```
steve@Eve:~$ source /opt/ros/foxy/setup.bash 
^[[A^[[Asteveipython3
Python 3.8.2 (default, Apr 27 2020, 15:53:34) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.13.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import cv_bridge                                                        
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
RuntimeError: FATAL: module compiled as little endian, but detected different endianness at runtime
---------------------------------------------------------------------------
SystemError                               Traceback (most recent call last)
<ipython-input-1-19d934114051> in <module>
----> 1 import cv_bridge

/opt/ros/foxy/lib/python3.8/site-packages/cv_bridge/__init__.py in <module>
      4 # This try is just to satisfy doc jobs that are built differently.
      5 try:
----> 6     from cv_bridge.boost.cv_bridge_boost import cvtColorForDisplay, getCvType
      7 except ImportError:
      8     pass

SystemError: initialization of cv_bridge_boost raised unreported exception

```

So this appears to be a genuine issue

SteveMacenski, 2020-06-12:
Upstream master doesn't seem to exhibit, so I think it was already fixed and foxy just needs to be updated potentially.

mjcarroll, 2020-06-26:
I managed to get a little further and surface this error in GDB:

```
RuntimeError: FATAL: module compiled as little endian, but detected different endianness at runtime
```

Which doesn't make a ton of sense, given that x86 compile-time and runtime should both be little endian.

mjcarroll, 2020-06-26:
It looks like it may be potentially addressed by #292

jacobperron, 2020-07-09:
See https://github.com/ros-perception/vision_opencv/pull/346 for a fix.

mjcarroll, 2020-07-28:
Closed by #342

closed 2020-07-28 by mjcarroll

togaen, 2020-11-25:
In case it helps anyone else: I just ran into this error, but doing `apt install python3-opencv` seems to have resolved whatever the problem was.

asimonov, 2021-04-29:
I am seeing this issue on Noetic on ARM64 (Jetson NX)

I run ubuntu20 in docker and see this (with default cv_bridge)

```
# source /opt/ros/noetic/setup.bash 
# roscd cv_bridge
/opt/ros/noetic/share/cv_bridge# python
Python 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv_bridge
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/__init__.py", line 6, in <module>
    from cv_bridge.boost.cv_bridge_boost import cvtColorForDisplay, getCvType
SystemError: initialization of cv_bridge_boost raised unreported exception
>>> 
```

and if I take current master and build it in my workspace I see the same issue:

```
root@uc1-nx-1:~/git/catkin_ws# source devel/setup.bash 
root@uc1-nx-1:~/git/catkin_ws# roscd cv_bridge
root@uc1-nx-1:~/git/catkin_ws/src/vision_opencv/cv_bridge# python
Python 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'python' is not defined
>>> import cv_bridge
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tii/git/catkin_ws/devel/lib/python3/dist-packages/cv_bridge/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 6, in <module>
SystemError: initialization of cv_bridge_boost raised unreported exception
>>> 
```

Any ideas how to debug/fix?

asimonov, 2021-04-29:
also, there is this URL describing same error on ARM64: https://answers.ros.org/question/350904/cv_bridge-throws-boost-import-error-in-python-3-and-ros-melodic/, but is for Melodic. And it also lists no solution, but a workaround to not use cv_bridge functions.

IannGeorges, 2021-04-29:
I tried that last week by creating that exact function inside my own script
and also using the repo vision_opencv test/starter-code. I referenced it
directly inside the same program as opposed to importing it and it still
gave me errors on both occasions :( Nonetheless I appreciate you sharing
that with me, please feel free to keep sending me any links you find
helpful!

asimonov, 2021-05-04:
after rebuilding ROS Noetic from source on Jetson NX and still seeing the same issue I have found a workaround:

https://github.com/opencv/opencv/issues/14884#issuecomment-599852128

import cv2 has to happen first in python scripts. then 'import cv_bridge' works fine

PR #342 (2020-07-10): Change module.hpp to align with Noetic
This makes `module.hpp` match what is in the `noetic` version.

It should also fix #339 by correctly returning a `nullptr` from `do_numpy_import`.

The reason for this is explained in #292.