ros-perception/vision_opencv #341: Unable to find the requested Boost libraries : boost_python37

Joechencc, 2020-06-17:
I have error when boot  python library. When do I fix this bug?

CMake Error at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1947 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.65.1

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_python37

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
  Boost.
Call Stack (most recent call first):
  vision_opencv/cv_bridge/CMakeLists.txt:11 (find_package)

jacksonhshields, 2020-06-24:
I had the same problem. What ROS version are you using? The default branch is set to noetic, if you are using melodic, try checking out the melodic branch

Joechencc, 2020-06-26:
It is melodic. However the core is ARM64

chinedu0507, 2020-07-02:
I had the same issue (with noetic), I fixed it by commenting out lines 11 and 12 in the CMakeLists.txt file in the /home/pc_name/catkin_ws/src/vision_opencv/cv_bridge directory. I am running Pop OS 20.04 (based on Ubuntu 20.04 (Focal Fossa). 

I was able to successfully compile the package after that. I hope this helps.

doubleZ0108, 2021-01-27:
@chinedu0507 he said just right, you can comment out `find_package(Boost REQUIRED python37)` and use `find_package(Boost REQUIRED python)` below instead.
Then in my case I fix my error.
Hope this state why and help.

TheMaakarov, 2021-02-20:
> I had the same issue (with noetic), I fixed it by commenting out lines 11 and 12 in the CMakeLists.txt file in the /home/pc_name/catkin_ws/src/vision_opencv/cv_bridge directory. I am running Pop OS [...]

As an alternative, this worked for me in Ubuntu 18.04: 
https://answers.ros.org/question/344951/could-not-find-the-following-boost-libraries-boost_python3/

xela1601, 2022-07-03:
> @chinedu0507 he said just right, you can comment out `find_package(Boost REQUIRED python37)` and use `find_package(Boost REQUIRED python)` below instead. Then in my case I fix my error. Hope this state why and help.

this helped me resolving this, thank you.
Also you can use 
```bash
sed -i '11,12 s/^/#/' cv_bridge/CMakeLists.txt
```
to comment out these lines through sed. Very handy if you use Docker e.g.