ros-perception/vision_opencv #345: Project 'cv_bridge' can not find opencv

cnpcshangbo, 2020-07-06:
Hi,
I am compiling https://github.com/thien94/vision_to_mavros.
I have installed mavros and vision_opencv from apt command. However, Project 'cv_bridge' can not find opencv.
```
CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
  Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  vision_to_mavros/CMakeLists.txt:10 (find_package)
```
Do you know which environment variable should I set to let Project 'cv_bridge' find opencv?
Thanks,
Bo

asasine, 2020-07-07:
I am getting this same error when compiling https://github.com/IntelRealSense/realsense-ros

sh0w, 2020-07-24:
i have the same issue on jetson TX2 + ubuntu 18.04

gaotao19890725, 2020-08-09:
> i had the same problem (jetson xavier + ubuntu 18.04 + ros melodic)
> 
> we fixed it like this:
> 
> change the file:
> 
> `/opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake `
> change this line:
> 
> `set(_include_dirs "include;/usr/include;/usr/include/opencv") `
> to
> 
> `set(_include_dirs "include;/usr/include;/usr/include/opencv4")`

Thank you.

cnpcshangbo, 2020-08-25:
It seems this error is related to OpenCV on Jetson.

yyyll, 2020-11-24:
i had the same problem，but I don‘t solve it. so can you tell me how to resolve ir.

cnpcshangbo, 2020-11-24:
@yyyll I didn't solve this. I commented the cv_bridge related stuff from https://github.com/thien94/vision_to_mavros since I don't need that.

yyyll, 2020-11-25:
ok,I understand

KoaTheKoala, 2021-01-01:
@sh0w This fix worked for me as well on:
Nvidia Jetson TX2 with Jetpack 4.4.1 (Ubuntu 18.04) and ROS Melodic.

Thanks :)!

XinnWang, 2021-01-07:
Opencv4 is installed by Nvidia's jetson repo, which replaced the original opencv3.
The most simple way to solve this is to comment out all the repo inside /etc/apt/sources.list.d/nvidia-l4t-apt-source.list , then apt update and reinstall libopencv-dev. 
Check this [link](https://www.cnblogs.com/xiangzh/p/14244915.html) for help.

Captain299792458, 2022-02-11:
> i had the same problem (jetson xavier + ubuntu 18.04 + ros melodic)
>
> we fixed it like this:
>
> change the file:
>
> `/opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake ` change this line: [...]

This works! Thanks  a lot

lovesuiyue, 2022-03-02:
Project 'image_geometry' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the website
  'http://www.ros.org/wiki/image_geometry' for information and consider
  reporting the problem.

so , how do i solve this problem

itaouil, 2022-04-21:
> i had the same problem (jetson xavier + ubuntu 18.04 + ros melodic)
>
> we fixed it like this:
>
> change the file:
>
> `/opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake ` change this line: [...]

Worked :)

Thanks :)

Dhruti1994, 2022-11-07:
> Thank you.

But this file not allowing to edit, can you tell us how did you change ?

AhmedARadwan, 2022-12-21:
I faced the same issue on Jetson Xavier AGX. Thanks to @sh0w the problem is now solved.
If you want to quickly fix it in a dockerfile, add the following command before building the workspace:

``` bash
sed -i 's/set(_include_dirs "include;\/usr\/include;\/usr\/include\/opencv")/set(_include_dirs "include;\/usr\/include;\/usr\/include\/opencv4")/g' /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake
```

lisichen123, 2023-05-16:
> Project 'image_geometry' specifies '/usr/include/opencv' as an include dir, which is not found. It does neither exist as an absolute directory nor in '${{prefix}}//usr/include/opencv'. Check the [...]

i had same problem, how to fix it ? any help will be appreciate

Jawaheru, 2024-03-13:
> i had the same problem (jetson xavier + ubuntu 18.04 + ros melodic)
>
> we fixed it like this:
>
> change the file:
>
> `/opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake ` change this line: [...]

I changed it but still now working, this error shows
[image]

ijnek, 2024-03-15:
Maintainer here.
Could we leave a thumbs up on this comment if this issue still persists for active distros on ubuntu 20.04 and ubuntu22.04?

paaastras, 2024-11-08:
> > 我遇到了同样的问题（jetson xavier + ubuntu 18.04 + ros melodic）
> > 我们这样修复它：
> > 更改文件：
> > `/opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake `更改此行：
> > `set(_include_dirs "include;/usr/include;/usr/include/opencv") `到
> > `set(_include_dirs "include;/usr/include;/usr/include/opencv4")`
> 
> 我改变了它，但现在仍然有效，显示此错误 ![2024-03-13 15-22-55 的截图](https://private-user-images.githubusercontent.com/100447343/312454130-c029500a-4ff6-437e-aa1f-660b43da9ab7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEwNDUwNzIsIm5iZiI6MTczMTA0NDc3MiwicGF0aCI6Ii8xMDA0NDczNDMvMzEyNDU0MTMwLWMwMjk1MDBhLTRmZjYtNDM3ZS1hYTFmLTY2MGI0M2RhOWFiNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwOFQwNTQ2MTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ODlmMWYxMGJhMjU5NzRlMTM5MWE3M2Q0ZTE4NGQxN2E1OTU4ZWZmOGZkZDhmODdlYWNlN2FjOWM1ODVmN2I4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.OPfKjv2SVckVPZGSPoJx_Sw0HaYKTbvUM0vwpwvxPzc)

hello 
look this：
> > `set(_include_dirs "include;/usr/include;/usr/local/include/opencv4")`

leocencetti, 2025-08-25:
For ROS `melodic` is probably better to install OpenCV `v3.2`, instead of coercing the build system into using `v4.x` (which can result in compilation errors).

```bash
apt purge libopencv-dev
apt install libopencv-dev=3.2.*
```

This will pull the OpenCV stack from the Ubuntu Bionic PPA instead of the NVIDIA one.