ros-perception/vision_opencv #347: using cv_bridge with python 3.6 on ubunut 16.04

waspinator, 2020-07-14:
I need to use python 3.6 in a kinetic ROS package that manipulates images using opencv. To transfer images in and out of the package I'm using cv_bridge, but it is only supported on python 2.7 so I'm trying to figure out a way to get around that. I have seen reports of this being possible but I'm unable to reproduce the results. 

- https://github.com/pulver22/gym-gazebo/wiki/ERROR:-Cv_bridge-for-Python3
- https://medium.com/ai%C2%B3-theory-practice-business/use-gpu-in-your-pytorch-code-676a67faed09
- https://stackoverflow.com/questions/49221565/unable-to-use-cv-bridge-with-ros-kinetic-and-python3
- https://stackoverflow.com/questions/47859699/unable-to-import-cv2-and-cvbridge-in-ros-and-python3?noredirect=1&lq=1
- https://stackoverflow.com/questions/57168989/building-cv-bridge-package-with-ros-kinetic-and-python3-ignoring-cmake-arguments?noredirect=1&lq=1

Below is the dockerfile which I use to build my environment. Running it, starting a python3.6 shell and importing cv_bridge   

```python3
from cv_bridge.boost.cv_bridge_boost import getCvType
```

gives the following error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)
```

running `source /cv_bridge_ws/install/setup.bash` doesn't help. Strangely though if I run that exact same command another time then it seems to work.

How do I properly source the workspace to use cv_bridge?

```dockerfile
FROM ros:kinetic

RUN apt-get update && apt-get install -y \
    software-properties-common python-catkin-tools \
    curl wget python-dev build-essential checkinstall \
    libreadline-gplv2-dev libncursesw5-dev libssl-dev \
    libsqlite3-dev tk-dev libgdbm-dev libc6-dev libbz2-dev

RUN apt-get update && apt-get install -y \
    ros-kinetic-cv-bridge \
    ros-kinetic-image-transport \
    ros-kinetic-tf \
    ros-kinetic-diagnostic-updater \
    ros-kinetic-ddynamic-reconfigure

RUN add-apt-repository -y ppa:deadsnakes/ppa \
    && apt-get update && apt-get install -y python3.6-dev

RUN curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py && \
    python2 get-pip.py && python3.6 get-pip.py

RUN pip3 install opencv-python PyYAML cython rospkg

RUN ln -s /usr/lib/x86_64-linux-gnu/libboost_python-py35.so /usr/lib/x86_64-linux-gnu/libboost_python3.so
RUN mkdir --parents /cv_bridge_ws/src
WORKDIR /cv_bridge_ws
RUN catkin config \
    -DPYTHON_EXECUTABLE=/usr/bin/python3.6 \
    -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m \
    -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
RUN catkin config --install
RUN git -C /cv_bridge_ws/src clone -b 1.12.8 https://github.com/ros-perception/vision_opencv.git 
RUN catkin config --extend /opt/ros/kinetic && \
    catkin clean --yes && \
    catkin build

MasterKID223, 2020-12-19:
try `source /cv_bridge_ws/install/setup.bash --extend`

ijnek, 2024-11-17:
Cloosing due to inactivity, please comment to reopen

closed 2024-11-17 by ijnek