ros-perception/vision_opencv #348: Can´t include cv_bridge in ros:foxy docker container

ItsMeTheBee, 2020-07-20:
Hey there!

I followed these [instructions](https://index.ros.org/p/cv_bridge/github-ros-perception-vision_opencv/) and pulled, build and sourced vision_opencv in my workspace. I´m working in a ros:foxy docker container if that makes any difference.

I tried running `colcon test` which resulted in this error:
```
--- stderr: opencv_tests                    
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/usr/lib/python3/dist-packages/pytest.py", line 102, in <module>
    raise SystemExit(pytest.main())
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 63, in main
    config = _prepareconfig(args, plugins)
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 206, in _prepareconfig
    return pluginmanager.hook.pytest_cmdline_parse(
  File "/usr/lib/python3/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/lib/python3/dist-packages/pluggy/manager.py", line 92, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/lib/python3/dist-packages/pluggy/manager.py", line 83, in <lambda>
    self._inner_hookexec = lambda hook, methods, kwargs: hook.multicall(
  File "/usr/lib/python3/dist-packages/pluggy/callers.py", line 203, in _multicall
    gen.send(outcome)
  File "/usr/lib/python3/dist-packages/_pytest/helpconfig.py", line 94, in pytest_cmdline_parse
    config = outcome.get_result()
  File "/usr/lib/python3/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/lib/python3/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 739, in pytest_cmdline_parse
    self.parse(args)
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 947, in parse
    self._preparse(args, addopts=addopts)
  File "/usr/lib/python3/dist-packages/_pytest/config/__init__.py", line 893, in _preparse
    self.pluginmanager.load_setuptools_entrypoints("pytest11")
  File "/usr/lib/python3/dist-packages/pluggy/manager.py", line 297, in load_setuptools_entrypoints
    plugin = ep.load()
  File "/usr/lib/python3.8/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_testing/__init__.py", line 15, in <module>
    from . import tools
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_testing/tools/__init__.py", line 18, in <module>
    from .process import launch_process
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_testing/tools/process.py", line 18, in <module>
    import launch.actions
ModuleNotFoundError: No module named 'launch.actions'
```
I was unable to find anything concerning this error message and cv_bridge tests finished without any error so i just continued as normal.
After adding `find_package(cv_bridge REQUIRED) ` to my Cmakelist I tried including cv_bridge with `#include "cv_bridge/cv_bridge.h"`. 

After that wasn´t successful I tried it like this:` #include "../../vision_opencv/cv_bridge/include/cv_bridge/cv_bridge.h"` 
The include is successful but now i get the error message 

```
In file included from /home/dev_ws/src/nn_tflite/src/test_tflite_node.cpp:28:
/home/dev_ws/src/nn_tflite/src/../../vision_opencv/cv_bridge/include/cv_bridge/cv_bridge.h:43:10: fatal error: opencv2/core/core.hpp: No such file or directory
   43 | #include <opencv2/core/core.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~

```

Just for fun I modified the Cmakelists in cv_bridge and cv_bridge/src to include this line `find_package(OpenCV REQUIRED)`
And in cv_bridge/src/CMakeList I changed
`target_link_libraries(${PROJECT_NAME} ${Boost_LIBRARIES} ${OpenCV_LIBRARIES})`

to
`target_link_libraries(${PROJECT_NAME} ${Boost_LIBRARIES} ${OpenCV_LIBS})`

simply because thats how I include OpenCv in my project.

Now the error reads
```
/home/dev_ws/src/nn_tflite/src/../../vision_opencv/cv_bridge/include/cv_bridge/cv_bridge.h:49:10: fatal error: cv_bridge/cv_bridge_export.h: No such file or directory
   49 | #include <cv_bridge/cv_bridge_export.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
`
I searched for cv_bridge_export.h but I was unable to find it.

Now I´m out of ideas so I thought maybe you guys could help me out :smile: 

PS: I know my fiddeling around with the CMakeLists has probably been useless but maybe it helps figuring out why i can´t include cv_bridge the way it´s supposed to work.

sh0w, 2020-07-24:
i have the same problem (jetson TX2 + ubuntu 18.04 + ros melodic)

ijnek, 2022-09-12:
The ``cv_bridge_export.h`` file gets generated by cv_bridge during compilation, so you won't find it as a file in the source.

``#include <cv_bridge/cv_bridge.h>`` is the correct syntax, and you're missing an ``ament_target_dependencies`` (alternatively ``target_link_libraries``) call to link the cv_bridge library to your target.

As this issue was posted over 2 years ago, I'm going to assume you're not going to try out these suggestions and going to close this off. Feel free to comment if you want it reopened.

closed 2022-09-12 by ijnek