ros-perception/vision_opencv #349: Could NOT find Boost (missing: Boost_INCLUDE_DIR python3)

wkoa, 2020-07-28:
[image]
In my case, I have failed to execute "**colcon build"**
My environment is :

- MacOS Catalina

- ROS2 Foxy

- Cmake 3.18

- Boost 1.72

Sammysedo, 2020-09-15:
Did you get any progress on this? I'm stuck at the same issue with Windows 10-built foxy.

wkoa, 2020-09-16:
> Did you get any progress on this? I'm stuck at the same issue with Windows 10-built foxy.

Not yet

cardboardcode, 2021-04-29:
+1. Stuck at the same issue as well here.

cardboardcode, 2021-05-12:
@wkoa @Sammysedo 

Were you guys able to resolve this issue? Been attempting for a week now but to no avail.

Below are details of what I tried so far:

## Environment
1. Windows 10.0.19042 N/A Build 19042
2. Visual Studio Code 2019 Community Edition
3. Boost 1.71.0

## Installation of BOOST
1. Downloaded `boost_1_71_0.zip` from the official `boost.org` website.
2. Extracted to under the administrative C Drive folder `Program Files`.
3. Opened a Visual Studio Native Command Prompt.
4. Navigated to the extracted folder for BOOST 1.71.0 and ran the following command:
```cmd
cd C:\Program Files\boost\boost_1_71_0
bootstrap vc142
.\b2
```

I have looked at #289 which is resolved by an earlier merged pull request and have also verified that the installed BOOST library I have contains a `python` library folder at the following local directory:
> C:\Program Files\boost\boost_1_71_0\libs\python

The `Path` environment variable has also been verified to link to that specific library. Below is a terminal output of the following command:
```cmd
echo %Path% | findstr boost\boost_1_71_0\libs\python
```
#### Terminal Output
```cmd
C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\libnvvp;C:\Python38\Scripts\;C:\Python38\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\ProgramData\chocolatey\bin;C:\Program Files\OpenSSL-Win64\bin;C:\opencv\x64\vc16\bin;C:\ProgramData\chocolatey\lib\cunit\lib;C:\ProgramData\chocolatey\lib\tinyxml2\lib;C:\ProgramData\chocolatey\lib\log4cxx\lib;C:\ProgramData\chocolatey\lib\bullet\lib;C:\Program Files\Graphviz\bin;C:\Program Files\Git\cmd;C:\Program Files\Git\bin;C:\Users\beyhy\cd-comms;C:\Program Files\NVIDIA Corporation\Nsight Compute 2019.5.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Users\beyhy\anaconda3\Scripts;C:\Users\beyhy\AppData\Local\atom-beta\bin;C:\Program Files\VideoLAN\VLC;C:\Qt\5.15.2\mingw81_64\bin;C:\Program Files\Cppcheck;C:\Program Files (x86)\GnuWin32\bin;C:\iconv-1.14-win32-x86_64\bin;C:\zlib-1.2.8-win32-x86_64\bin;C:\xmllint\bin;C:\Python38\Lib\site-packages\PyQt5\Qt5\bin;C:\Program Files\CMake\bin;C:\Program Files\boost\boost_1_71_0;C:\Program Files\boost\boost_1_71_0\libs\python;;C:\Users\beyhy\AppData\Local\Microsoft\WindowsApps;C:\Users\beyhy\AppData\Local\atom-beta\bin
```

I have also tried replacing **line 32** of `CMakeLists.txt` in `cv_bridge` under the **ros2** branch from this:
```txt
find_package(Boost COMPONENTS python3 REQUIRED)
```
to this:
```txt
find_package(Boost COMPONENTS python REQUIRED)
```
Since the python library in the installed BOOST library is labelled `python` instead of `python3`.

However, that still did not work. 

## Appreciate any help or constructive feedback at the moment. Not sure how to proceed after hours of google searching for alternative solutions.

flynneva, 2021-05-12:
i ran into the same problem, and solved it by adding the following code to [line 32 in the CMakeList.txt](https://github.com/ros-perception/vision_opencv/blob/05062b20bb086104d3a45d0ac17bdb9b7be8bf20/cv_bridge/CMakeLists.txt#L32) file within the `cv_bridge` pacakge:

```
string(REPLACE "." ";" VERSION_LIST ${PYTHONLIBS_VERSION_STRING})
list(GET VERSION_LIST 0 PYTHONLIBS_VERSION_MAJOR)
list(GET VERSION_LIST 1 PYTHONLIBS_VERSION_MINOR)
find_package(Boost COMPONENTS python${PYTHONLIBS_VERSION_MAJOR}${PYTHONLIBS_VERSION_MINOR} REQUIRED)
```

flynneva, 2021-05-12:
im working on a PR now to add this update into the main `ros2` branch

cardboardcode, 2021-05-13:
@flynneva 
Thanks for the prompt clarification 😄 as well as the pull request to fix this long-standing issue. 

I have tried the solution you provided and the results are similar, as shown below;

## What I Did
Followed the modifications made in the pull request #390. 
See [here](https://github.com/ros-perception/vision_opencv/pull/390/commits/2e59c59da567a1d639f54e1b5a29b26c68a47453) for what the **cv_bridge** CMakeLists.txt files looks after modifying.

I have ensured no other unintentional modifications are made by redownloading the **ros2** branch of **vision_opencv** via the following commands:
```cmd
rem Opening up the Visual Studio Native Command Prompt Terminal
cd C:\Users\beyhy\ros_workspaces\vcam_ws\src
git clone https://github.com/ros-perception/vision_opencv.git --single-branch --branch ros2 --depth 1
cd  C:\Users\beyhy\ros_workspaces\vcam_ws
call C:\opt\ros\foxy\local_setup.bat
colcon build --packages-select cv_bridge
```

## Results
Similar error as before. See screenshot below.
[image]

It continues to generate the error: 
```cmd
Could NOT find Boost (missing: python38) (found version "1.71.0")
```

## Further Debugging
I have verified my Python installation as shown below by running `python` in the Visual Studio Native Command Prompt:
[image]

I don't think **Boost.Python** is using its own internal PythonLibs so I am unsure what else I am doing wrong here.

flynneva, 2021-05-13:
@cardboardcode do you have `boost-python` installed? you can figure this out by navigating to where you installed boost and scrolling through the `libs` directory for python. if its there, great you have it (and you should be able to see the python version you need to specify in your CMakeList.txt then too!)

its not included in the pure header library provided by boost, you either have to download the [pre-compiled version of boost](https://sourceforge.net/projects/boost/files/boost-binaries/) or compile it yourself. id recommend downloading the pre-compiled version and adding it to your environment variable (you need to add the full path to where-ever you installed boost to your `PATH` environment variable.

So for instance on my window setup I added this to my `PATH`: `C:\boost\boost_1_76_0\lib64-msvc-14.2`

yours might be different depending on what boost version you install. i've spent my fair share of grinding through getting boost installed on windows so let me know if you have issues still....and know that I feel your pain.

flynneva, 2021-05-13:
i think going forward it would be best to remove the boost dependency all together from this package...but that'll take some work id expect. ill eventually look into it, just dont have the time now

cardboardcode, 2021-05-13:
Thanks @flynneva. I finally managed to resolve this build error and successfully built `cv_bridge` with what you have suggested above. 

I will delve into deeper details on how I resolved it on my side shortly.

cardboardcode, 2021-05-13:
## Source Of Issue
Looking at what resolved it for me personally, the **reasons why I encountered the build error** is due to **2 sources**:

1. Ambiguous referencing of **BOOST.python** under [line 32 of cv_bridge CMakeLists.txt](https://github.com/ros-perception/vision_opencv/blob/05062b20bb086104d3a45d0ac17bdb9b7be8bf20/cv_bridge/CMakeLists.txt#L32)
2. Incompatibility between `Python 3.8` and `BOOST 1.71.0`

### Why did I have **Python 3.8**?
In following the [Installing ROS2 on Windows](https://docs.ros.org/en/foxy/Installation/Windows-Install-Binary.html#install-python) article from the official ROS2 Foxy documentation, the instructions recommended installing **python 3.8** via the command:
```cmd
choco install -y python --version 3.8.3
```
### Why did I have **BOOST 1.71.0**?
This was the first BOOST library version I came across which is verified to be compatible and stable with Visual Studio 2019 Community Edition. As stated under **Solution**, I would later realize that **BOOST 1.71.0** is only compatible with **Python 3.7.**

## [🙌Solution🙌]
This section covers how the build error is resolved by addressing the 2 aformentioned sources:

### Installing precompiled BOOST via `.msi` 
1. Go to the Downloads page from [the official BOOST website](https://sourceforge.net/projects/boost/files/boost-binaries/1.71.0/) and download whichever bit-specific installer you need. 
> Eg. For me, I got `boost_1_71_0-msvc-12.0-64.exe`.
2. Run the installer.
3. Add your specific BOOST library path to the **Windows 10 system environment variable**, `Path`. 
> Eg. I added `C:\local\boost_1_71_0\lib64-msvc-14.2` to `Path`.

### Finding out What Python Library version is compatible with BOOST 1.71.0
1. Go to where you have downloaded and extracted your BOOST library installation. Eg. For me, my BOOST library was in the following directory: 
```txt
C:\local\boost_1_71_0
```
2. Go to the `lib64-msvc-14.2` sub-folder and use the explorer windows to search for this keyword `libboost-python`. You should get something similar to below:
[image]

With that, I identified that this version of BOOST library requires **Python 3.7** instead **Python 3.8** which I have. 

### Changing Python Library version
1. Use the **Add or Remove Programs** Windows program to remove your current version of Python.
2. Go to the official **[Python Release Downloads](https://www.python.org/downloads/windows/)** page and get [Python 3.7.9](https://www.python.org/ftp/python/3.7.9/python-3.7.9-amd64.exe). 
3. Run the installer. 
4. Remove the previous Python library from the `Path` system environment variable.
> Eg. I removed C:\Python38\ and C:\Python38\Scripts
5. Include the new Python library into the `Path` system environment variable. 
> Eg. I included C:\Python37\ and C:\Python37\Scripts

### Reinstalling ROS2 Python Dependencies.
In deleting **Python 3.8**, all your python dependencies you installed before for ROS2 will be gone. So you will have to install it again via the following commands:
```cmd
python -m pip install -U catkin_pkg cryptography empy ifcfg lark-parser lxml netifaces numpy opencv-python pyparsing pyyaml setuptools
python -m pip install -U pydot PyQt5
```

### Verifying That It Works
Go through the steps below to verify if the build error is still encountered. 

```cmd
rem Opening up the Visual Studio Native Command Prompt Terminal
mkdir -p c:\ros2_ws\src
cd c:\ros2_ws\src
git clone https://github.com/flynneva/vision_opencv.git --single-branch --branch ros2 --depth 1
cd c:\ros2_ws\
call c:\opt\ros\foxy\local_setup.bat
colcon build --packages-select cv_bridge
```

You should get an output similar to the one below: 
[image]

## And done. Phew. **Selamat Hari Raya** and hope this helps whoever is facing this same build error on Windows for ROS2 Foxy.

cardboardcode, 2021-05-13:
@flynneva 

Thanks for the help in resolving this issue and prevent further hair from growing white due to stress. 😅 

I will test your pull request in a MacOS environment as well and address the original build error encountered by @wkoa, ... once I have time. 

Once that is done, we can safely close this issue.

cardboardcode, 2021-05-14:
# [ 🙌 Solution 🙌 ]

Hi all, please follow this instead. Realized the solution I proposed above is faulty because using **Python 3.7.9** instead of **Python 3.8.3** will cause ROS2 commands to fail. 

So here's the rectified version of the instructions. 

## Install Dependencies
1. Python 3.8.3
2. BOOST 1.73.0

### **Python 3.8.3**
For this, just use the recommended command in the ROS2 documentation in an **administrator command prompt**:
```cmd
choco install -y python --version 3.8.3
```

### **BOOST 1.73.0**
Identified that BOOST 1.73.0 is compatible with Python 3.8 using the instructions in [Finding out What Python Library version is compatible with BOOST 1.71.0](https://github.com/ros-perception/vision_opencv/issues/349#issuecomment-840351488).

1. Go to [the Downloads page](https://sourceforge.net/projects/boost/files/boost-binaries/1.73.0/) from the official BOOST website and download whichever bit-specific installer you need.

> Eg. For me, I got boost_1_73_0-msvc-12.0-64.exe.

2. Run the installer.
3. Add your specific BOOST library path to the Windows 10 system environment variable, `Path`.
> Eg. I added C:\local\boost_1_73_0\lib64-msvc-14.2 to Path.

## Verifying That It Works
Go through the steps below to verify if the build error is still encountered.
```cmd
rem Opening up the Visual Studio Native Command Prompt Terminal
mkdir -p c:\ros2_ws\src
cd c:\ros2_ws\src
git clone https://github.com/flynneva/vision_opencv.git --single-branch --branch ros2 --depth 1
cd c:\ros2_ws\
call c:\opt\ros\foxy\local_setup.bat
colcon build --packages-select cv_bridge
```

shimanStone, 2022-11-10:
sudo -H apt-get install -y libboost-dev

ijnek, 2024-11-17:
Issue is explained in detail, and there seems to be no actionables for this repository. Closing!

closed 2024-11-17 by ijnek

GPROWIRELESS, 2024-12-17:
Well I got the same issue again after I followed the instructions from @cardboardcode.
The software versions are:
windows11
msvs14
Ros2 Humble
Vision_opencv-humble
python3.8.3
boost pre-complied 1.7.4

cmake] CMake Error at C:/Program Files/CMake/share/cmake-3.30/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
[cmake]   Could NOT find Boost (missing: python3) (found version "1.74.0")
[cmake] Call Stack (most recent call first):
[cmake]   C:/Program Files/CMake/share/cmake-3.30/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
[cmake]   C:/Program Files/CMake/share/cmake-3.30/Modules/FindBoost.cmake:2409 (find_package_handle_standard_args)
[cmake]   CMakeLists.txt:55 (find_package)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
[proc] The command: "C:\Program Files\CMake\bin\cmake.EXE" -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE --no-warn-unused-cli -SC:/ros2/ros_ws/src/vision_opencv-humble/cv_bridge -Bc:/ros2/ros_ws/src/vision_opencv-humble/cv_bridge/build -G "Visual Studio 16 2019" -T host=x86 -A x64 exited with code: 1