ros-perception/vision_opencv #364: Multiple undefined references in libboost_python3.so

m-morelli, 2020-09-14:
Hello,
I'm on Debian 10, where I've installed libboost-python1.67-dev. I need to build cv_bridge as a dependency for [navigation2](https://navigation.ros.org/build_instructions/index.html#build-navigation2-dependencies) (foxy version). I have ros2 foxy installed from sources in my machine and properly sourced in the terminal where I do build cv_bridge.

Below [1] is what I get. Any idea of how to solve?

Thanks in advance.
Matteo

[1]
~/nav2_depend_ws$ colcon build --symlink-install 
Starting >>> camera_calibration_parsers
Starting >>> cv_bridge
Finished <<< camera_calibration_parsers [0.37s]                                      
Starting >>> bond
--- stderr: cv_bridge                                                                  
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyUnicode_InternFromString'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyLong_FromLong'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyModule_Create2'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_InPlaceRemainder'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PySlice_New'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyList_Sort'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyObject_SetAttrString'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyType_IsSubtype'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyList_Insert'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyExc_RuntimeError'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyComplex_Type'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyDict_Copy'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyEval_CallFunction'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyDict_Size'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyUnicode_Type'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyStaticMethod_New'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyList_Reverse'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyErr_NoMemory'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `_Py_fopen'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyUnicode_FromString'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyBaseObject_Type'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyList_Type'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyModule_Type'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_InPlaceLshift'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyBytes_AsString'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyDict_Keys'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_Subtract'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyObject_DelItem'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_InPlaceMultiply'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `_PyType_Lookup'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyErr_Clear'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyArg_ParseTupleAndKeywords'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyCallable_Check'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_Xor'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyLong_AsUnsignedLongLong'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyErr_ExceptionMatches'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyMethod_New'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyBytes_Size'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_Or'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyLong_AsSsize_t'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_InPlaceOr'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyExc_TypeError'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyComplex_RealAsDouble'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyErr_SetObject'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyLong_AsUnsignedLong'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyType_Ready'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyObject_IsTrue'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_Add'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyTuple_GetItem'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyDict_Clear'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyObject_Size'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyObject_RichCompare'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyLong_AsLongLong'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyBool_Type'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_Multiply'
[... 62 lines cut ...]
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyObject_CallMethod'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_InPlaceAnd'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyObject_GetAttrString'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyMem_Malloc'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyList_New'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyErr_Format'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyObject_SetAttr'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyNumber_InPlaceXor'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyCFunction_NewEx'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyLong_FromUnsignedLong'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyStaticMethod_Type'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libboost_python3.so: undefined reference to `PyTuple_Size'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/cv_bridge-utest.dir/build.make:197: test/cv_bridge-utest] Error 1
make[1]: *** [CMakeFiles/Makefile2:290: test/CMakeFiles/cv_bridge-utest.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
---
Failed   <<< cv_bridge [0.56s, exited with code 2]
Aborted  <<< bond [0.37s]                                

Summary: 1 package finished [1.02s]
  1 package failed: cv_bridge
  1 package aborted: bond
  1 package had stderr output: cv_bridge
  17 packages not processed

nplan, 2020-09-27:
@m-morelli
I have the same issue, also runing Debian 10 (arm) with libboost-python-1.67-dev installed. Did you figure it out?

svengit, 2020-09-28:
@nplan @m-morelli 
I had the same issue, explicitly linking to python3.7 fixed it for me:

    colcon build --symlink-install --packages-select image_transport cv_bridge --cmake-args "-DCMAKE_CXX_STANDARD_LIBRARIES=-lpython3.7m" --cmake-force-configure

svengit, 2020-09-30:
Short follow-up:
Adding  `${PYTHON_LIBRARIES}` to ` target_link_libraries(${PROJECT_NAME}-utest)` here: https://github.com/ros-perception/vision_opencv/blob/2026f6d66833e0830f9ce69004d43c9abd1e5fa4/cv_bridge/test/CMakeLists.txt#L24 
properly fixes this problem.

nplan, 2020-10-04:
@svengit 
I tried your solutions and both of them fixed the problem. Thanks. Maybe you should open a pull request.

jdcdp, 2020-10-07:
See #318 . I had the same problem but as of today it does not happen any more. Rebuilding ros2 and vision_opencv gives no compilation errors. Could you check whether the issue was solved?

akshitj1, 2020-11-28:
@jdcdp Same issue occured with me, while building on fedora. I am now able to build successfully by using fix on #318. ie.
`target_link_libraries(${PROJECT_NAME} ${PYTHON_LIBRARIES} ${Boost_LIBRARIES} ${OpenCV_LIBRARIES})`
I think #318 may have been closed prematurely.

ijnek, 2024-11-17:
This was fixed in #370

closed 2024-11-17 by ijnek