ros-perception/vision_opencv #366: colcon build failing on MacOS Catalina 10.15.3

ashutoshshrm529, 2020-10-24:
I am trying to build this package with gazebo_ros_packages for ROS version Foxy Fritzoy, on MacOS Catalina 10.15.3. 
The build fails for the package cv_bridge with the following error message - 

```
--- stderr: cv_bridge                                                                                             
Undefined symbols for architecture x86_64:
  "cv::error(int, cv::String const&, char const*, char const*, int)", referenced from:
      cv::Mat::Mat(int, int, int, void*, unsigned long) in cv_bridge.cpp.o
  "cv::String::deallocate()", referenced from:
      cv::String::~String() in cv_bridge.cpp.o
      cv::String::operator=(cv::String const&) in cv_bridge.cpp.o
      cv::String::~String() in rgb_colors.cpp.o
      cv::String::operator=(cv::String const&) in rgb_colors.cpp.o
  "cv::String::allocate(unsigned long)", referenced from:
      cv::String::String(char const*) in cv_bridge.cpp.o
      cv::String::String(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in cv_bridge.cpp.o
  "cv::imencode(cv::String const&, cv::_InputArray const&, std::__1::vector<unsigned char, std::__1::allocator<unsigned char> >&, std::__1::vector<int, std::__1::allocator<int> > const&)", referenced from:
      cv_bridge::CvImage::toCompressedImageMsg(sensor_msgs::msg::CompressedImage_<std::__1::allocator<void> >&, cv_bridge::Format) const in cv_bridge.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/libcv_bridge.dylib] Error 1
make[1]: *** [src/CMakeFiles/cv_bridge.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< cv_bridge [3.99s, exited with code 2]
```

Any idea why this is happening, and possible fixes? Would really appreciate any help in this regard.

ijnek, 2022-09-12:
Hi @ashutoshshrm529 , just checking up to see if this issue is still active?

ashutoshshrm529, 2022-09-12:
Hi @ijnek, I'm not certain if the issue is still active, I've since switched platforms and am currently using the Linux version of ROS on a virtual machine. 

Although for anyone who's trying this - I do remember that compiling this library from scratch instead helped on Mac OS Catalina.

I'll mark this issue as closed. Thanks for bringing that to my attention.

closed 2022-09-12 by ashutoshshrm529

ijnek, 2022-09-12:
@ashutoshshrm529 thanks for providing an update!