ros-perception/vision_opencv #367: cv_bridge is unable to find opencv

JohnVorwald, 2020-12-31:
Here is the error message when running make ros for MYNT-EYE-S-SDK on Ubuntu 18.06.

~/MYNT-EYE-S-SDK$ make -j6 ros
...
Make ros
Base path: /home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros
Source space: /home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/src
Build space: /home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/build
Devel space: /home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/devel
Install space: /home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/install
####
#### Running command: "cmake /home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/devel -DCMAKE_INSTALL_PREFIX=/home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/install -G Unix Makefiles" in "/home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/devel
-- Using CMAKE_PREFIX_PATH: /home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/devel;/opt/ros/melodic
-- This workspace overlays: /home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - mynt_eye_ros_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'mynt_eye_ros_wrapper'
-- ==> add_subdirectory(mynt_eye_ros_wrapper)
CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
  Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  mynt_eye_ros_wrapper/CMakeLists.txt:36 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/johnvorwald/Github/MYNT-EYE-S-SDK/wrappers/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Here are the locations of "opencv" in /usr/include

johnvorwald@johnvorwald-desktop:~/MYNT-EYE-S-SDK$ find /usr/include -name *pencv* -print
/usr/include/boost/compute/interop/opencv.hpp
/usr/include/boost/compute/interop/opencv
/usr/include/opencv4
/usr/include/opencv4/opencv2
/usr/include/opencv4/opencv2/opencv_modules.hpp
/usr/include/opencv4/opencv2/opencv.hpp
/usr/include/opencv4/opencv2/gapi/opencv_includes.hpp

How should I fix this?

MartensCedric, 2021-01-01:
I have the same problem for an Nvidia jetson TX2

KoaTheKoala, 2021-01-01:
I have the same problem for an Nvidia jetson TX2

CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
  Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  cv_camera/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/Programming/robotics-prototype/robot/rospackages/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/Programming/robotics-prototype/robot/rospackages/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

MartensCedric, 2021-01-01:
Duplicate for #345 which fixed the problem for me

ijnek, 2022-09-12:
Closing, as duplicate of #345

closed 2022-09-12 by ijnek