ros-perception/vision_opencv #369: ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)

Hiroaki-K4, 2021-01-06:
Hi!
When I built my program with catkin build, I got the following error:
```
File "/opt/ros/melodic/lib/python2.7/dist-packages/cv_bridge/core.py", line 91, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)
```
After searching for the cause, I found that cv_bridge is built with python2 by default.
My environment is as follows.
・ Virtualenv (python3.6)
・ ROS melodic
・ Jetson AGX xavier

I wanted to use cv_bridge with python3, so I used the link below to build cv_bridge with the following steps.
https://stackoverflow.com/questions/49221565/unable-to-use-cv-bridge-with-ros-kinetic-and-python3

```
sudo apt-get install python-catkin-tools python3-dev python3-catkin-pkg-modules python3-numpy python3-yaml ros-kinetic-cv-bridge
# Create catkin workspace
mkdir catkin_ws
cd catkin_ws
catkin init
# Instruct catkin to set cmake variables
catkin config -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/aarch64-linux-gnu/libpython3.6m.so
# Instruct catkin to install built packages into install place. It is $CATKIN_WORKSPACE/install folder
catkin config --install
# Clone cv_bridge src
git clone https://github.com/ros-perception/vision_opencv.git src/vision_opencv
# Find version of cv_bridge in your repository
apt-cache show ros-kinetic-cv-bridge | grep Version
cd src/vision_opencv/
git checkout 1.13.0(maybe, I forgot)
cd ../../
# Build
catkin build cv_bridge
# Extend environment with new package
source install/setup.bash --extend
```
After performing this procedure, from cv_bridge.boost.cv_bridge_boost import getCvType no longer causes an error.
After that, I put other nodes I created in catkin_ws and built it with catkin build. However, when I run it with roslauch, I get the following error:
```
RLException: [sample.launch] is neither a launch file in package [sample_proc] nor is [sample_proc] a launch file name
```
Running source devel / setup.bash does not eliminate this error.
Do I need to do anything different if I use virtualenv? Thanks!

closed 2021-01-07 by Hiroaki-K4

bryansgue, 2022-04-08:
https://cyaninfinite.com/ros-cv-bridge-with-python-3/

bryansgue, 2022-04-08:
works for me

Buttosaii, 2022-11-30:
Hi, when i tried to 'source install/setup.bash --extend', it shows 'bash: install/setup.bash: No such file or directory'. Am I doing any step wrongly? I am using a venv running python3.6.

Thank you!