ros-perception/vision_opencv #372: how to include libpython to make the unit tests?

nosense345, 2021-01-23:
Hi,

I'm trying to get vision_opencv to work with ROS2-Foxy on Arch Linux.

I'm almost there ( I 've had to add #define BOOST_BIND_GLOBAL_PLACEHOLDERS to src/module.hpp to get rid of a deprecated warning) but I'm stuck on the unit tests.
I can get the unit test to compile by adding /usr/lib/libpython3.9.so.1.0 to  
test/CMakeFiles/cv_bridge-utest.dir/build.make:
[...]
test/cv_bridge-utest: /usr/lib/libboost_python39.so.1.75.0
test/cv_bridge-utest: /usr/lib/libpython3.9.so.1.0   //  <- added 
[...]

However, that change is reset during each build, and therefore I don't know how to get it integrated with ROS2
Where/how can I make this change permanent? I don't know much about cmake or colcon

Thanks for your help,
Klaas

ijnek, 2022-09-12:
Hi @nosense345, just checking up to see if this issue is still relevant or it can be closed?

ijnek, 2024-11-17:
Closing due to inactivity