ros-perception/vision_opencv #376: '/home/agx/catkin_ws/src/vision_opencv-noetic/cv_bridge/include' as an  include dir, which is not found.

afridi26, 2021-02-04:
Hey,

I try to run the catkin_make command for [def_cam_teledyne_nano-refactor_code](https://github.com/Brazilian-Institute-of-Robotics/def_cam_teledyne_nano) package. I manually downloaded vision_opencv form this repo as instructed [here](https://github.com/cipherdev/src/issues/13#issuecomment-552785165) in #13 but later I deleted it, before and now its the same I am getting the following error:

```
-- ==> add_subdirectory(def_cam_teledyne_nano-refactor_code)
CMake Error at /home/agx/catkin_ws/devel/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
  Project 'cv_bridge' specifies
  '/home/agx/catkin_ws/src/vision_opencv-noetic/cv_bridge/include' as an
  include dir, which is not found.  It does neither exist as an absolute
  directory nor in
  '/home/agx/catkin_ws/src/vision_opencv-noetic/cv_bridge//home/agx/catkin_ws/src/vision_opencv-noetic/cv_bridge/include'.
  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  def_cam_teledyne_nano-refactor_code/CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/agx/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/agx/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:2294: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1

```
I searched alot and not able to fixed this issue.  I have also looked at this issue #367 and #345 . But no success. I am new to Ros and not able to resolve this issue. Any help would be appreciated.

Thanks

Yan-Xiaodi, 2021-03-05:
I have met this question too,hope someone can tell me how to fixed that.I have tried the solution in #345 but it's not worked for me.

closed 2021-03-10 by afridi26

wesen02, 2023-02-08:
how do you solve this?