ros-perception/vision_opencv #385: cv_bridge issues with opencv_contrib

linzha0, 2021-04-20:
Hi, I am using Ubuntu 20.04 and ROS noetic, OpenCV 4.2 is installed by ROS. I am trying to use xfeature2d from opencv_contrib. then I build Opencv 4.2 and opnecv_contib 4.2 from source in order to use cv_bridge. In my project CMakelists.txt, I inlcude the new installed Opencv like below, all the output is correct:  
```cmake
    SET(OpenCV_DIR /home/lin/develop/3rd/opencv/install/opencv-4.2.0/lib/cmake/opencv4/)   
    find_package(OpenCV 4.2)    
    message(STATUS "OpenCV library status:")  
    message(STATUS "    config: ${OpenCV_DIR}")  
    message(STATUS "    version: ${OpenCV_VERSION}")  
    message(STATUS "    libraries: ${OpenCV_LIBS}")  
    message(STATUS "    include path: ${OpenCV_INCLUDE_DIRS}")
```
**Outputs:**

> -- OpenCV library status:
> --     config: /home/lin/develop/3rd/opencv/install/opencv-4.2.0/lib/cmake/opencv4/
> --     version: 4.2.0
> --     libraries: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_gapi;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_stitching;opencv_video;opencv_videoio;opencv_aruco;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_cvv;opencv_datasets;opencv_dnn_objdetect;opencv_dnn_superres;opencv_dpm;opencv_face;opencv_freetype;opencv_fuzzy;opencv_hdf;opencv_hfs;opencv_img_hash;opencv_line_descriptor;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_quality;opencv_reg;opencv_rgbd;opencv_saliency;opencv_sfm;opencv_shape;opencv_stereo;opencv_structured_light;opencv_superres;opencv_surface_matching;opencv_text;opencv_tracking;opencv_videostab;opencv_viz;opencv_xfeatures2d;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto
> --     include path: /home/lin/develop/3rd/opencv/install/opencv-4.2.0/include/opencv4
> 

Also, I set LD_LIBARARY_PATH and PATH like below:
```shell
export  PATH=/home/lin/develop/3rd/opencv/install/opencv-4.2.0/bin/:$PATH
export LD_LIBRARY_PATH=/home/lin/develop/3rd/opencv/install/opencv-4.2.0/lib/:$LD_LIBRARY_PATH
```

But when I build my project with **cv_bridge**, it always link the lib like `/usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.4.2.0`. I found /opt/ros/noetic/share/cv_bridge/cmake/cv_bridgeConfig.cmake, they were set the lib in `/usr/lib/x86_64-linux-gnu/`.

May I ask the cv_bridge will always link the lib from `/usr/lib/x86_64-linux-gnu/`? If that's the case, Do I have to rebuild the cv_bridge with my new installed opencv? Or maybe I miss some setup. 
Please give some suggestions. Thanks for your time.

adamanov, 2022-01-03:
hello, I have a similar problem, how could you solve it?

linzha0, 2022-01-03:
I build opencv 4.2, opencv_contrib, cv_bridge(build using opencv 4.2 in a new catkin work space) in a local path, still keep the original one. then link them in my cmakelist like this:

``` cmake
SET(OpenCV_DIR /home/lin/develop/3rd/opencv/install/opencv-4.2.0/lib/cmake/opencv4/)
find_package(OpenCV 4.2.0 REQUIRED)
message(STATUS "OpenCV library status:")
message(STATUS "    config: ${OpenCV_DIR}")
message(STATUS "    version: ${OpenCV_VERSION}")
message(STATUS "    libraries: ${OpenCV_LIBS}")
message(STATUS "    include path: ${OpenCV_INCLUDE_DIRS}")

SET(cv_bridge_DIR /home/lin/develop/ros/ros_ws/devel/share/cv_bridge/cmake/)
find_package(catkin REQUIRED COMPONENTS
   cv_bridge 
)
```

closed 2022-01-03 by linzha0