ros-perception/vision_opencv #386: catkin_make error -> boost python 3.7 libraries

IannGeorges, 2021-04-20:
OS: Ubuntu Mate 18.04 running on RPI 3 B+
ROS: ROS Melodic
Objective: I am trying to use cv_bridge in-order to use opencv on my image topic for ROS. I'm relatively new to ROS, let alone Linux/Ubuntu so I've been having trouble getting past this issue for a while. Whenever I use catkin_make I get errors from vision_opencv stemming from the cv_bridge. 

```
-- +++ processing catkin package: 'cv_bridge'
-- ==> add_subdirectory(vision_opencv/cv_bridge)
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython2.7.so (found version "2.7.17")
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python37; skipping header check
Call Stack (most recent call first):
  vision_opencv/cv_bridge/CMakeLists.txt:11 (find_package)

CMake Error at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1947 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.65.1

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_python37

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
  Boost.
Call Stack (most recent call first):
  vision_opencv/cv_bridge/CMakeLists.txt:11 (find_package)

-- Did not find OpenCV 4, trying OpenCV 3
-- Found OpenCV: /usr (found suitable version "3.2.0", minimum required is "3") found components:  opencv_core opencv_imgproc opencv_imgcodecs
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython2.7.so (found suitable version "2.7.17", minimum required is "2.7")
```

I believe the boost libraries are already installed, with command `find /usr/lib -iname 'libboost_python*.so*'` 
the output I get is: 
/usr/lib/aarch64-linux-gnu/libboost_python-py27.so
/usr/lib/aarch64-linux-gnu/libboost_python3.so
/usr/lib/aarch64-linux-gnu/libboost_python3-py36.so
/usr/lib/aarch64-linux-gnu/libboost_python.so
/usr/lib/aarch64-linux-gnu/libboost_python3-py36.so.1.65.1
/usr/lib/aarch64-linux-gnu/libboost_python-py36.so
/usr/lib/aarch64-linux-gnu/libboost_python-py27.so.1.65.1

I feel like the underlying issue might be related to directory location however I am still pretty new to all this stuff so I have no idea, if someone could assist me I'd greatly appreciate it. Thank you for time

IannGeorges, 2021-04-20:
I've also checked to see if boost is installed
```
$ dpkg -s libboost-dev | grep Version
Version: 1.65.1.0ubuntu1 
```

poett1, 2021-05-12:
same error here. Any suggestions?

flynneva, 2021-05-12:
from your logs it looks like you have `Python 2.7` installed, but boost is looking for boost `python37`.

either upgrade your python installation to 3.7 or change the CMake file to look for boost `python27`

kjwoo31, 2022-04-01:
This worked for me in melodic (from [link](https://answers.ros.org/question/344951/could-not-find-the-following-boost-libraries-boost_python3/))

```
cd /usr/lib/x86_64-linux-gnu
sudo ln -s libboost_python3.a libboost_python37.a
sudo ln -s libboost_python3.so libboost_python37.so
```

ok-kewei, 2022-08-26:
> Edit vision_opencv/cv_bridge/CMakeLists.txt , Change line no 11 from find_package(Boost REQUIRED python37) to find_package(Boost REQUIRED python)
> 
> After doing so , it worked for me

Life saver! <3

Mechazo11, 2022-12-31:
> This worked for me in melodic (from [link](https://answers.ros.org/question/344951/could-not-find-the-following-boost-libraries-boost_python3/)). If it already exists, then remove it with sudo rm -rf and try again.
> 
> ```
> cd /usr/lib/x86_64-linux-gnu
> sudo ln -s libboost_python3.a libboost_python37.a
> sudo ln -s libboost_python3.so libboost_python37.so
> ```

@kjwoo31 using your commands says file already exists. What to do from here?

kjwoo31, 2022-12-31:
> 

There are some instructions in the attached link. I just removed the existing files with sudo rm -rf command and tried again.

ijnek, 2024-11-17:
Closing due to inactivity, and since melodic has reached EOL.

closed 2024-11-17 by ijnek