ros-perception/vision_opencv #389: Project ‘cv_bridge‘ specifies ‘/usr/include/opencv‘ as an include dir, which is not found

cjl19721, 2021-05-09:
When I compile iai_Kinect v2, this error appears.The following is the error log.

> CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
  Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  iai_kinect2/kinect2_bridge/CMakeLists.txt:24 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/q/test_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/q/test_ws/build/CMakeFiles/CMakeError.log".
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

sienuc, 2021-05-14:
cd /usr/include/ 
sudo ln -s opencv4/ opencv (replace "4" with your version)

Had the same issue, the above fixed my compiler needs. 
Be ware of compatibility with yours, you may need to delete the link afterwards to not break other cmake runs that use/expect other versions of opencv. 

"Problem" is the static / hardcoded libraries in cv_bridge cmake file.

cjl19721, 2021-05-16:
Thank you
I have solved this problem.When I was installing opencv, I chose the home folder,So it caused this error.After I modified the path, it was no problem