ros-perception/vision_opencv #393: foxy and galactic branches?

flynneva, 2021-06-18:
i noticed that there are dashing and eloquent branches for this repo but no foxy or galactic. should those be created so active ros2 development doesn't break code for previous releases? or is the plan to try and make the `ros2` branch backwards compatible with all future ros2 releases?

ijnek, 2022-07-23:
Seems like the source entries in rosdistro all point to the ``ros2`` branch, but it seems like there is no attempt at making the branch backwards compatible, which is problematic.

I believe all active distros: ``foxy, ``galactic``, ``humble`` and ``rolling``should be separated to allow bug fix backporting.

ijnek, 2022-09-04:
I've [recently become the maintainer](https://github.com/ros-perception/vision_opencv/pull/451) of this repo, so I hope we can get through issues like this and get this repo back active.

``foxy``, ``galactic`` and ``humble`` have been created, and we can close this issue once https://github.com/ros/rosdistro/pull/34314 is merged.

ijnek, 2022-09-06:
Closing, as https://github.com/ros/rosdistro/pull/34314 is merged.

closed 2022-09-06 by ijnek