ros-perception/vision_opencv #401: [Error] Windows Build issue with BOOST

ShivamPR21, 2021-07-26:
Hi! all
I am new to windows environment with ROS ecosystem.

Here facing problems with windows build of `cv_bridge` stating
```shell
: fatal error LNK1120: 3 unresolved externals 
```

I have tried some of the open PRs for fixes, but that didn't work #390 

here is complete output: 
```shell
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363.
-- The C compiler identification is MSVC 19.28.29334.0
-- The CXX compiler identification is MSVC 19.28.29334.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake_ros: 0.9.2 (C:/dev/ros2-windows/share/ament_cmake_ros/cmake)
-- Found PythonInterp: C:/Python38/python.exe (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: C:/Python38/python.exe
-- Found PythonLibs: C:/Python38/libs/python38.lib (found version "3.8.10") 
-- Found Boost: C:/Program Files/PCL 1.12.0/3rdParty/Boost/lib/cmake/Boost-1.76.0/BoostConfig.cmake (found version "1.76.0")  
-- Found Boost: C:/Program Files/PCL 1.12.0/3rdParty/Boost/lib/cmake/Boost-1.76.0/BoostConfig.cmake (found version "1.76.0") found components: python 
-- Found sensor_msgs: 2.2.3 (C:/dev/ros2-windows/share/sensor_msgs/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
-- Found rosidl_adapter: 2.2.1 (C:/dev/ros2-windows/share/rosidl_adapter/cmake)
-- Found OpenSSL: optimized;C:/Python38/Library/lib/libcrypto.lib;debug;C:/Program Files/OpenSSL-Win64/lib/VC/libcrypto64MDd.lib (found version "1.1.1k")  
-- Found FastRTPS: C:/dev/ros2-windows/include  
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
-- Found OpenCV: C:/Users/hpd4kor/Desktop/opencv/build (found suitable version "4.5.2", minimum required is "4") found components: opencv_core opencv_imgproc opencv_imgcodecs 
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Failed
-- Performing Test COMPILER_HAS_DEPRECATED
-- Performing Test COMPILER_HAS_DEPRECATED - Success
-- Found PythonInterp: C:/Python38/python.exe (found version "3.8.10") 
-- Found PythonLibs: C:/Python38/libs/python38.lib (found suitable version "3.8.10", minimum required is "3.8") 
-- Found ament_lint_auto: 0.10.6 (C:/dev/ros2-windows/share/ament_lint_auto/cmake)
-- Found gtest sources under 'C:/dev/ros2-windows/src/gtest_vendor': C++ tests using 'Google Test' will be built
-- Configuring done
-- Generating done
-- Build files have been written to: C:/dev/carla-nav/build/cv_bridge
Microsoft (R) Build Engine version 16.8.2+25e4d540b for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Building Custom Rule C:/dev/carla-nav/src/vision_opencv/cv_bridge/python/CMakeLists.txt
  running egg_info
  creating cv_bridge.egg-info
  writing cv_bridge.egg-info\PKG-INFO
  writing dependency_links to cv_bridge.egg-info\dependency_links.txt
  writing top-level names to cv_bridge.egg-info\top_level.txt
  writing manifest file 'cv_bridge.egg-info\SOURCES.txt'
  reading manifest file 'cv_bridge.egg-info\SOURCES.txt'
  writing manifest file 'cv_bridge.egg-info\SOURCES.txt'
  Building Custom Rule C:/dev/carla-nav/src/vision_opencv/cv_bridge/python/CMakeLists.txt
  Building Custom Rule C:/dev/carla-nav/src/vision_opencv/cv_bridge/src/CMakeLists.txt
  cv_bridge.cpp
  rgb_colors.cpp
C:\dev\carla-nav\src\vision_opencv\cv_bridge\include\cv_bridge/cv_bridge.h(55,1): warning C4275: non dll-interface class 'std::runtime_error' used as base for dll-interface class 'cv_bridge::Exception' (compiling source file C:\dev\carla-nav\src\vision_opencv\cv_bridge\src\cv_bridge.cpp) [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.28.29333\include\stdexcept(107): message : see declaration of 'std::runtime_error' (compiling source file C:\dev\carla-nav\src\vision_opencv\cv_bridge\src\cv_bridge.cpp) [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge.vcxproj]
C:\dev\carla-nav\src\vision_opencv\cv_bridge\include\cv_bridge/cv_bridge.h(54): message : see declaration of 'cv_bridge::Exception' (compiling source file C:\dev\carla-nav\src\vision_opencv\cv_bridge\src\cv_bridge.cpp) [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge.vcxproj]
C:\dev\carla-nav\src\vision_opencv\cv_bridge\include\cv_bridge/cv_bridge.h(86,31): warning C4251: 'cv_bridge::CvImage::header': struct 'std_msgs::msg::Header_<std::allocator<void>>' needs to have dll-interface to be used by clients of class 'cv_bridge::CvImage' (compiling source file C:\dev\carla-nav\src\vision_opencv\cv_bridge\src\cv_bridge.cpp) [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge.vcxproj]
[... 77 lines cut ...]
  module_opencv3.cpp
C:\Program Files\PCL 1.12.0\3rdParty\Boost\include\boost-1_76\boost/bind.hpp(41): message : The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior. [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge_boost.vcxproj]
C:\Program Files\PCL 1.12.0\3rdParty\Boost\include\boost-1_76\boost/bind.hpp(41): message : The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior. [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge_boost.vcxproj]
C:\dev\carla-nav\src\vision_opencv\cv_bridge\src\module_opencv3.cpp(19,3): warning C4996: '_vsnprintf': This function or variable may be unsafe. Consider using _vsnprintf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge_boost.vcxproj]
C:\dev\carla-nav\src\vision_opencv\cv_bridge\src\module_opencv3.cpp(83,3): warning C5030: attribute 'gnu::unused' is not recognized [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge_boost.vcxproj]
C:\dev\carla-nav\src\vision_opencv\cv_bridge\src\module_opencv3.cpp(89,3): warning C4996: '_vsnprintf': This function or variable may be unsafe. Consider using _vsnprintf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge_boost.vcxproj]
     Creating library C:/dev/carla-nav/build/cv_bridge/src/Release/cv_bridge_boost.lib and object C:/dev/carla-nav/build/cv_bridge/src/Release/cv_bridge_boost.exp
LINK : warning LNK4217: symbol '?throw_error_already_set@python@boost@@YAXXZ (void __cdecl boost::python::throw_error_already_set(void))' defined in 'libboost_python38-vc142-mt-x64-1_76.lib(errors.obj)' is imported by 'module.obj' in function '"struct _object * __cdecl boost::python::detail::manage_ptr<struct _object>(struct _object *,...)" (??$manage_ptr@U_object@@@detail@python@boost@@YAPEAU_object@@PEAU3@ZZ)' [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge_boost.vcxproj]
LINK : warning LNK4217: symbol '??1?$handle@U_object@@@python@boost@@QEAA@XZ (public: __cdecl boost::python::handle<struct _object>::~handle<struct _object>(void))' defined in 'libboost_python38-vc142-mt-x64-1_76.lib(function_doc_signature.obj)' is imported by 'module.obj' in function '"public: __cdecl boost::python::detail::keyword::~keyword(void)" (??1keyword@detail@python@boost@@QEAA@XZ)' [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge_boost.vcxproj]
LINK : warning LNK4217: symbol '?lookup@registry@converter@python@boost@@YAAEBUregistration@234@Utype_info@34@@Z (struct boost::python::converter::registration const & __cdecl boost::python::converter::registry::lookup(struct boost::python::type_info))' defined in 'libboost_python38-vc142-mt-x64-1_76.lib(registry.obj)' is imported by 'module.obj' in function '"void __cdecl `dynamic  [... cut ...] ...)" (??$make_function1@P6A?AVobject@api@python@boost@@V1234@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@1@Z@detail@python@boost@@YA?AVobject@api@12@P6A?AV3412@V3412@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@1@ZZZ) [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge_boost.vcxproj]
C:\dev\carla-nav\build\cv_bridge\src\Release\cv_bridge_boost.dll : fatal error LNK1120: 3 unresolved externals [C:\dev\carla-nav\build\cv_bridge\src\cv_bridge_boost.vcxproj]
```

flynneva, 2021-07-26:
It looks like you are trying to use Boost 1.76 but you have Python 3.8 - which the latest version of boost that uses python 3.8 is 1.74.

Either upgrade your python version or downgrade your boost version and you should be good.

The error is because the boost version you have installed was built with Python 3.9.

flynneva, 2021-07-26:
i'd recommend [using the link in the PR you linked to above to download a pre-compiled Boost 1.74](https://sourceforge.net/projects/boost/files/boost-binaries/1.74.0/). once you do and set your proper environmental variables it should work.

wolfv, 2022-06-17:
I am actually facing the same issue with Boost 1.74 and Python 3.9. This used to work in Galactic, but for some reason seems to break in Humble. Trying to figure out what's going on!

wolfv, 2022-06-17:
You can find the linking error on our Windows build in Github actions: https://github.com/RoboStack/ros-humble/runs/6868272365?check_suite_focus=true Any ideas? Would be very much appreciated!

wolfv, 2022-06-24:
I fixed this by adding this to src/CMakeLists.txt

```
if (MSVC)
    target_compile_definitions(${PROJECT_NAME}_boost PRIVATE -DBOOST_PYTHON_STATIC_LIB)
endif()
```

SENAI-GilmarCorreia, 2024-08-19:
Hello ROS developers! I'm encountering the same error mentioned by @ShivamPR21 when building this package on Windows. I'm using Boost 1.74.0, Python 3.8, and OpenCV 4.1.

```bash
module.obj : error LNK2019: s�mbolo externo n�o resolvido, "__declspec(dllimport) public: __cdecl boost::python::handle<struct _object>::handle<struct _object>(void)" (__imp_??0?$handle@U_object@@@python@boost@@QEAA@XZ), referenciado na fun��o "public: void __cdecl boost::python::detail::keyword::`default constructor closure'(void)" (??_Fkeyword@detail@python@boost@@QEAAXXZ) [C:\Desktop\naad_ws\build\cv_bridge\src\cv_bridge_boost.vcxproj]
module.obj : error LNK2019: s�mbolo externo n�o resolvido, "__declspec(dllimport) public: __cdecl boost::python::handle<struct _object>::handle<struct _object>(class boost::python::handle<struct _object> const &)" (__imp_??0?$handle@U_object@@@python@boost@@QEAA@AEBV012@@Z), referenciado na fun��o "public: __cdecl boost::python::detail::keyword::keyword(struct boost::python::detail::keyword &&)" (??0keyword@detail@python@boost@@QEAA@$$QEAU0123@@Z) [C:\Desktop\naad_ws\build\cv_bridge\src\cv_bridge_boost.vcxproj]
module.obj : error LNK2019: s�mbolo externo n�o resolvido, "__declspec(dllimport) public: __cdecl boost::python::objects::py_function_impl_base::py_function_impl_base(void)" (__imp_??0py_function_impl_base@objects@python@boost@@QEAA@XZ), referenciado na fun��o "class boost::python::api::object __cdecl boost::python::detail::make_function1<class boost::python::api::object (__cdecl*)(class boost::python::api::object,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)>(class boost::python::api::object (__cdecl*)(class boost::python::api::object,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &),...)" (??$make_function1@P6A?AVobject@api@python@boost@@V1234@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@1@Z@detail@python@boost@@YA?AVobject@api@12@P6A?AV3412@V3412@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@1@ZZZ) [C:\Desktop\naad_ws\build\cv_bridge\src\cv_bridge_boost.vcxproj]
C:\Desktop\naad_ws\build\cv_bridge\src\boost\Release\cv_bridge_boost.pyd : fatal error LNK1120: 3 externo n�o resolvidos [C:\Desktop\naad_ws\build\cv_bridge\src\cv_bridge_boost.vcxproj]
```

How can I resolve this issue?

SENAI-GilmarCorreia, 2024-08-20:
I'm not sure how I solved it, but adding the @wolfv part at iron branch and rebooting the PC worked. ¯\_(ツ)_/¯

PR #538 (2024-12-02): Add Windows Compatibility Fixes for vision_opencv Build
### Summary

This PR adds compatibility improvements to enable successful building of `vision_opencv` on Windows by introducing additional compiler options. 

### Changes Made

1. **Added Compiler Definitions for Windows Compatibility**:
   - Added `-DHAVE_SNPRINTF` to `add_compile_options` to ensure that the `snprintf` function is available during compilation. This is a common requirement when building cross-platform code on Windows.

2. **Boost Python Library Static Linking on Windows**:
   - For MSVC (Microsoft Visual C++), `target_com [... cut]