ros-perception/vision_opencv #404: colcon build of cv_bridge fails without '--symlink-install'

Algorhythm-sxv, 2021-08-25:
when running `sudo colcon build --packages-select cv_bridge` without the option `--symlink-install`, the following error occurs:

```
--- stderr: cv_bridge
CMake Error at CMakelists.txt:88 (install):
  install TARGETS given target "cv_bridge" which does not exist in this
  directory.

---
```

the command `sudo colcon build --symlink-install --packages-select cv_bridge` completes without errors.

System details:
NVIDIA Jetson Xavier AGX
Ubuntu 18.04 (Jetpack 4.6.1)
OpenCV 3.2.0 (downgraded from the default OpenCV 4)
ROS 2 built from source

flynneva, 2021-08-25:
make sure you clean (remove your build and install directories completely) your colcon ws each time you switch between plain `colcon build` and then using `--symlink-install`.  that error seems like you went from using `--symlink-install` to not and then back again.

I could build the `ros2` branch for foxy just fine on my machine:

```
colcon build --packages-select cv_bridge
Starting >>> cv_bridge
Finished <<< cv_bridge [10.4s]                       

Summary: 1 package finished [10.5s]
```

ijnek, 2022-09-12:
Closing, as the previous comment provides a suggestion and the original author has not confirmed whether it has worked or not. Feel free to comment on this issue if it is still relevant, and we can consider getting it re-opened.

closed 2022-09-12 by ijnek