ros-perception/vision_opencv #405: cmake_make "Makefile:140: recipe for target 'all' failed"

MehmetcanKaymaz, 2021-09-03:
when I try to build, I get this error

desktop:~/catkin_ws$ catkin_make 
Base path: /home/deep-drone/catkin_ws
Source space: /home/deep-drone/catkin_ws/src
Build space: /home/deep-drone/catkin_ws/build
Devel space: /home/deep-drone/catkin_ws/devel
Install space: /home/deep-drone/catkin_ws/install
####
#### Running command: "cmake /home/deep-drone/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/deep-drone/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/deep-drone/catkin_ws/install -G Unix Makefiles" in "/home/deep-drone/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/deep-drone/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/deep-drone/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 4 packages in topological order:
-- ~~  - opencv_tests
-- ~~  - vision_opencv (metapackage)
-- ~~  - cv_bridge
-- ~~  - image_geometry
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'opencv_tests'
-- ==> add_subdirectory(vision_opencv/opencv_tests)
-- +++ processing catkin metapackage: 'vision_opencv'
-- ==> add_subdirectory(vision_opencv/vision_opencv)
-- +++ processing catkin package: 'cv_bridge'
-- ==> add_subdirectory(vision_opencv/cv_bridge)
CMake Warning at vision_opencv/cv_bridge/CMakeLists.txt:5 (find_package):
  By not providing "FindPython3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Python3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Python3" with any
  of the following names:

    Python3Config.cmake
    python3-config.cmake

  Add the installation prefix of "Python3" to CMAKE_PREFIX_PATH or set
  "Python3_DIR" to a directory containing one of the above files.  If
  "Python3" provides a separate development package or SDK, be sure it has
  been installed.

-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython2.7.so (found version "2.7.17") 
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python37; skipping header check
Call Stack (most recent call first):
  vision_opencv/cv_bridge/CMakeLists.txt:13 (find_package)

-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   python37
-- Found OpenCV: /usr/local (found suitable version "3.4.1", minimum required is "3") found components:  opencv_core opencv_imgproc opencv_imgcodecs 
-- Found OpenCV: /usr/local (found version "3.4.1") found components:  opencv_calib3d opencv_core opencv_cudaarithm opencv_cudabgsegm opencv_cudacodec opencv_cudafeatures2d opencv_cudafilters opencv_cudaimgproc opencv_cudalegacy opencv_cudaobjdetect opencv_cudaoptflow opencv_cudastereo opencv_cudawarping opencv_cudev opencv_dnn opencv_features2d opencv_flann opencv_highgui opencv_imgcodecs opencv_imgproc opencv_ml opencv_objdetect opencv_photo opencv_shape opencv_stitching opencv_superres opencv_video opencv_videoio opencv_videostab 
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython2.7.so (found suitable version "2.7.17", minimum required is "2.7") 
-- +++ processing catkin package: 'image_geometry'
-- ==> add_subdirectory(vision_opencv/image_geometry)
-- Found OpenCV: /usr/local (found version "3.4.1") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/deep-drone/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/deep-drone/catkin_ws/build"
####
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_nodejs
[ 33%] Built target image_geometry
[ 66%] Built target cv_bridge
[ 88%] Building CXX object vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module.cpp.o
[ 88%] Building CXX object vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module_opencv4.cpp.o
In file included from /usr/include/python2.7/numpy/ndarrayobject.h:27:0,
                 from /home/deep-drone/catkin_ws/src/vision_opencv/cv_bridge/src/module.hpp:26,
                 from /home/deep-drone/catkin_ws/src/vision_opencv/cv_bridge/src/module_opencv4.cpp:3:
/home/deep-drone/catkin_ws/src/vision_opencv/cv_bridge/src/module.hpp: In function ‘void* do_numpy_import()’:
/usr/include/python2.7/numpy/__multiarray_api.h:1537:144: error: return-statement with no value, in function returning ‘void*’ [-fpermissive]
 #define import_array() {if (_import_array() < 0) {PyErr_Print(); PyErr_SetString(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } }
                                                                                                                                                ^
/usr/include/python2.7/numpy/__multiarray_api.h:1537:144: note: in definition of macro ‘import_array’
 #define import_array() {if (_import_array() < 0) {PyErr_Print(); PyErr_SetString(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } }
                                                                                                                                                ^~~~~~
In file included from /usr/include/python2.7/numpy/ndarrayobject.h:27:0,
                 from /home/deep-drone/catkin_ws/src/vision_opencv/cv_bridge/src/module.hpp:26,
                 from /home/deep-drone/catkin_ws/src/vision_opencv/cv_bridge/src/module.cpp:35:
/home/deep-drone/catkin_ws/src/vision_opencv/cv_bridge/src/module.hpp: In function ‘void* do_numpy_import()’:
/usr/include/python2.7/numpy/__multiarray_api.h:1537:144: error: return-statement with no value, in function returning ‘void*’ [-fpermissive]
  PyErr_SetString(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } }
                                                                                ^
/usr/include/python2.7/numpy/__multiarray_api.h:1537:144: note: in definition of macro ‘import_array’
  PyErr_SetString(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } }
                                                                                ^~~~~~
vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/build.make:86: recipe for target 'vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module_opencv4.cpp.o' failed
make[2]: *** [vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module_opencv4.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/build.make:62: recipe for target 'vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module.cpp.o' failed
make[2]: *** [vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module.cpp.o] Error 1
CMakeFiles/Makefile2:1040: recipe for target 'vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/all' failed
make[1]: *** [vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

I am not sure is the problem python

MehmetcanKaymaz, 2021-09-07:
The solution is [https://github.com/ros-perception/vision_opencv/issues/329#issuecomment-635192230](url)

closed 2021-09-07 by MehmetcanKaymaz

adityachechani, 2022-11-22:
@MehmetcanKaymaz The link doesn't seem to be working. Can you please post a working solution link? Thanks!