ros-perception/vision_opencv #406: [ros2] colcon build failed with cv_bridge

northdk, 2021-09-08:
I'm trying build ros2-galactic on my Ubuntu-18.04, I got error like this:

`--- stderr: cv_bridge                                                                                                                                                      
CMake Error at CMakeLists.txt:88 (install):
  install TARGETS given target "cv_bridge" which does not exist in this
  directory.

make: *** [cmake_check_build_system] Error 1
---
Failed   <<< cv_bridge [3.37s, exited with code 2]
Aborted  <<< rcl_lifecycle [1.61s]                   `

northdk, 2021-09-08:
with temp solution, I remove the install code in [CMakeLists.txt](https://github.com/ros-perception/vision_opencv/blob/ros2/cv_bridge/CMakeLists.txt)

`install(TARGETS ${PROJECT_NAME}
        RUNTIME DESTINATION bin
        ARCHIVE DESTINATION lib
        LIBRARY DESTINATION lib
)
`

northdk, 2021-09-08:
my working branch is ros2

sloretz, 2022-03-24:
First, Galactic targets Ubuntu Focal (20.04) not Bionic (18.04): https://www.ros.org/reps/rep-2000.html#galactic-geochelone-may-2021-november-2022 . There doesn't seem to be a `galactic` specific branch, so one should be created and that branch should be set here: https://github.com/ros/rosdistro/blob/f712a1136e2ae498dbab2e04b57689e8101f2a82/galactic/distribution.yaml#L5561 .

If you want to build the latest release from source, checkout the tag that was last released to ROS Galactic: https://github.com/ros-perception/vision_opencv/tree/2.2.1

If you'd like to make the branch work on Ubuntu Bionic (18.04), this post suggests moving the install rule into the `src/CMakeLists.txt` file could work.
https://stackoverflow.com/questions/34443128/cmake-install-targets-in-subdirectories

ijnek, 2022-09-12:
I'm going to close this due to inactivity. Please comment on this issue if it is still relevant, to get it re-opened.

closed 2022-09-12 by ijnek