ros-perception/vision_opencv #415: error while compiling cv_bridge

2019new, 2021-10-30:
hi,i am using cv_bridge in my ros package, but i came up with an error whiling compiling it:
/opt/ros/melodic/include/cv_bridge/cv_bridge.h:43:10: fatal error: opencv2/core/core.hpp NO such file or directory
it seems that unable to include file <opencv2/core/core.hpp>
However, find_package(OpenCV 4 REQUIRED) succeeded to find OpenCV, value of ${OpenCV_INCLUDE_DIRS} is usr/local/include/opencv4. can anyone tell me what's wrong with it.
I am using NVIDA AGX XAVIER. Thanks
[CMakeLists.txt](https://github.com/ros-perception/vision_opencv/files/7445751/CMakeLists.txt)

wxmerkt, 2021-12-13:
We are running into a similar issue on an Nvidia NX. Our include directory is `/usr/include/opencv4` but the binary specifies `/usr/include/opencv`:
```
CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
  Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:9 (find_package)
```

kovacsge11, 2022-01-10:
Maybe this would be useful for somebody who faces the same problem: for me the problem was that I forgot to add cv_bridge as an ament_target_dependency to my target in my package where I used it.

wangruoyao, 2022-06-17:
> 

So, could you make it more detailed? I met the same problem on win10
C:\dev\ros2_foxy\vision_opencv-ros2>colcon build --symlink-install
Starting >>> cv_bridge
Starting >>> image_geometry
--- stderr: cv_bridge
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findament_cmake_ros.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_ros", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake_ros"
  with any of the following names:

    ament_cmake_rosConfig.cmake
    ament_cmake_ros-config.cmake

  Add the installation prefix of "ament_cmake_ros" to CMAKE_PREFIX_PATH or
  set "ament_cmake_ros_DIR" to a directory containing one of the above files.
  If "ament_cmake_ros" provides a separate development package or SDK, be
  sure it has been installed.

---
Failed   <<< cv_bridge [4.97s, exited with code 1]
--- stderr: image_geometry
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findament_cmake_python.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_python", but CMake did not find one.

  Could not find a package configuration file provided by
  "ament_cmake_python" with any of the following names:

    ament_cmake_pythonConfig.cmake
    ament_cmake_python-config.cmake

  Add the installation prefix of "ament_cmake_python" to CMAKE_PREFIX_PATH or
  set "ament_cmake_python_DIR" to a directory containing one of the above
  files.  If "ament_cmake_python" provides a separate development package or
  SDK, be sure it has been installed.

ijnek, 2022-09-12:
Closing due to inactivity, and the last coment https://github.com/ros-perception/vision_opencv/issues/415#issuecomment-1158518266 is now covered by #438. Please comment if the issue **in the original post** is still present, and if this needs to be re-opened.

closed 2022-09-12 by ijnek

tdqcb, 2023-08-27:
Could not find a package configuration file provided by "ament_cmake_ros"
with any of the following names:

ament_cmake_rosConfig.cmake
ament_cmake_ros-config.cmake

I find I didn't chose the correct version when i met this problem. if you use git clone to acquire the code,the error is  easy to occur.my ros is noetic ,so when i down the code ,I need to chose the noetic.Hope this is help for you.