ros-perception/vision_opencv #426: ROS2 Foxy: In ubuntu 20.04 Focal cv_bridge fails w/ not finding python3-dev

dirksavage88, 2022-03-05:
Environment: Ros2 foxy, 20.04.4 LTS (Focal Fossa)
Problem: Colcon build fails when trying to build cv_bridge
"--- stderr: cv_bridge                                                                                                   
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find Python3 (missing: Python3_LIBRARIES Python3_INCLUDE_DIRS
  Python3_NumPy_INCLUDE_DIRS Development NumPy) (found version "3.8.10")"

Troubleshooting: It looks like the Focal security updates upgraded python3.8.2 to python3.8.10, but only python3-dev=3.8.2-0ubuntu2 500 is a suitable python3-dev candidate according to apt cache policy as seen below:
[image]

So there is a mismatch between the python3.8 version and python3-dev version that breaks cmake paths?

Omkar-IMP, 2022-05-12:
Hey! were you able to fix it! or have any workarounds for this?

Thank you!

dirksavage88, 2022-05-12:
Hello, I think we can close this as I figured out I needed to uninstall python 2. 

I upgraded from ubuntu 1804 to 2004 and the python 2 libraries that remained in the file system (and it’s default paths) created a conflict with Python 3.8 and this package. Even update alternatives to prioritize python3 did not fix the issue.

 I would not say this is a fix per se but ubuntu upgrades come at the cost of conflicting python versions and this issue trickles down into ROS packages.

Omkar-IMP, 2022-05-13:
Thank you!!
I was able to build it

closed 2022-05-13 by dirksavage88