ros-perception/vision_opencv #429: Link cv_bridge with opencv 3.4.15

Snow-know-nothing, 2022-03-27:
I use the noetic branch of vision_opencv. Because I want to use opencv 3.4.15, I modify the CMakeLists.txt like following:
[image]
Then I came up with the error when I complile it 
```
`-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /bin/cc
-- Check for working C compiler: /bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/ubuntu/vision_opencv-noetic/cv_bridge/build/devel
-- Using CMAKE_PREFIX_PATH: /home/ubuntu/catkin_ws/devel;/opt/ros/noetic
-- This workspace overlays: /home/ubuntu/catkin_ws/devel;/opt/ros/noetic
-- Found PythonInterp: /bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ubuntu/vision_opencv-noetic/cv_bridge/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /bin/python3 (found version "3.8.10") 
-- Found Threads: TRUE  
-- Using Python nosetests: /bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython3.8.so (found version "3.8.10") 
-- Found Boost: /lib/aarch64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: python 
-- Found OpenCV: /usr/local (found suitable version "3.4.15", minimum required is "3") found components: opencv_core opencv_imgproc opencv_imgcodecs 
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython3.8.so (found suitable version "3.8.10", minimum required is "3.8") 
-- Configuring done
CMake Error in src/CMakeLists.txt:
  Imported target "Boost::python" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in src/CMakeLists.txt:
  Imported target "Boost::python" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
`
```
I have no idea what should I do
 I worked in :

- Ubuntu 20.04 on Raspberry Pi 
- ros noetic and I have removed the original cv_bridge

19020022037, 2022-04-20:
According to the following https://github.com/VowpalWabbit/vowpal_wabbit/issues/3003, which was referenced [here](https://stackoverflow.com/questions/70585114/imported-target-boostsystem-includes-non-existent-path-include), you need to create a symlink for /include -> /usr/include. To do so run the following command in the Terminal:
sudo ln -s /usr/include /include

ijnek, 2022-09-12:
@Snow-know-nothing Did you want to keep this open? Did the work around suggested work?