ros-perception/vision_opencv #431: [cv_bridge] Is it correct check for color format?

standmit, 2022-04-01:
https://github.com/ros-perception/vision_opencv/blob/c791220cefd0abf02c6719e2ce0fea465857a88e/cv_bridge/src/cv_bridge.cpp#L203-L206
There are both checks for color format and mono format in the same place. Is it a bug?

ijnek, 2022-09-12:
I belive what its doing here, is it's checking whether the source and destination encoding is a valid ("known") encoding type, rather than checking whether the source and destination is a "color" (ie. not grey) format. So I don't think it's a bug.

galou, 2024-05-06:
I also [stumbled  upon this](https://github.com/ros-perception/vision_opencv/issues/431). @ijnek Are you sure about this?

galou, 2025-05-21:
If the checking is here to express that the format is know, then the error message of the exception should be changed accordingly.

I have an issue when converting from a `16UC1` sensor_msgs/msg/Image to `MONO16`  CV Mat. The error message in the exception is obviously wrong: `[16UC1] is not a color format. but [MONO16] is. The conversion does not make sense`.

ijnek, 2025-06-14:
I understand the concern with the error message, it is kind of misleading. What about having
```cpp
throw Exception("Cannot convert from [" + src_encoding + "] to [" + dst_encoding +
                "]: source encoding is not a known image format (e.g. mono or color), so conversion is semantically invalid.");
```
instead?

Conversion from 16UC1 to MONO16, should be achievable via a call like this
```cpp
cv_bridge::toCvCopy(msg, "mono16")->image
// or
cv_bridge::toCvCopy(msg, sensor_msgs::image_encodings::MONO16)->image
```