ros-perception/vision_opencv #434: libcv_bridge.so：undefined reference to‘boost::re_detail_106000::perl_matcher<char const*, std::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::_match_flags)’

chengbutangls, 2022-05-11:
I rebuild the cv_bridge with opencv3.4.0. When I ran the "catkin_make" to build cv_bridge only ,the cv_bridge is ok.But when I  call the cv_bridge in my program，it has a problem that :
`/home/zzq/catkin_workspace/devel/lib/libcv_bridge.so：undefined reference to‘boost::re_detail_106000::perl_matcher<char const*, std::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::_match_flags)’`

My program that find the cv_bridge in CmakeLists.txt will be seen as follows:
```
#set(cv_bridge_DIR /home/zzq/catkin_ws/devel/share/cv_bridge/cmake)
find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  std_msgs
  sensor_msgs
  cv_bridge
)
```
The first line of the program is not work，so I add the “#”.

ijnek, 2022-09-12:
Hi @chengbutangls, I'm the maintainer of this repo. Just checking up if this issue still exists, or you've managed to find the problem?

zhulf0804, 2022-12-26:
Hello @ijnek, the issue still exists.

ijnek, 2022-12-27:
Thanks @zhulf0804 for confirming, I'll keep the issue open and will review related PRs.

tugbakara, 2023-01-05:
Is there any improvement about it, I have the same.

ijnek, 2023-01-08:
Not yet. I'm not involved in development of ROS1 branches, but happy to volunteer to review and merge PRs that fix this issue.

Potics-jz, 2023-05-24:
Is there have any improvement?
[image]
i have the same problem