ros-perception/vision_opencv #438: ros2 branches cannot be built on windows10

wangruoyao, 2022-06-16:
I tried to build it in win10 but failed.
I'm using ros2-foxy.

C:\dev\ros2_foxy\vision_opencv-ros2>git checkout ros2
fatal: not a git repository (or any of the parent directories): .git

C:\dev\ros2_foxy\vision_opencv-ros2>colcon build --symlink-install
Starting >>> cv_bridge
Starting >>> image_geometry
--- stderr: cv_bridge
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findament_cmake_ros.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_ros", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake_ros"
  with any of the following names:

    ament_cmake_rosConfig.cmake
    ament_cmake_ros-config.cmake

  Add the installation prefix of "ament_cmake_ros" to CMAKE_PREFIX_PATH or
  set "ament_cmake_ros_DIR" to a directory containing one of the above files.
  If "ament_cmake_ros" provides a separate development package or SDK, be
  sure it has been installed.

---
Failed   <<< cv_bridge [4.97s, exited with code 1]
--- stderr: image_geometry
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findament_cmake_python.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_python", but CMake did not find one.

  Could not find a package configuration file provided by
  "ament_cmake_python" with any of the following names:

    ament_cmake_pythonConfig.cmake
    ament_cmake_python-config.cmake

  Add the installation prefix of "ament_cmake_python" to CMAKE_PREFIX_PATH or
  set "ament_cmake_python_DIR" to a directory containing one of the above
  files.  If "ament_cmake_python" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< image_geometry [4.97s, exited with code 1]

Summary: 0 packages finished [5.28s]
  2 packages failed: cv_bridge image_geometry
  2 packages had stderr output: cv_bridge image_geometry
  2 packages not processed

wangruoyao, 2022-06-17:
after amend CMakelist and install boost. I build the image_geometry successfully but still cannot build cv_bridge without any information.

C:\ros2_ws\my_package\vision_opencv-ros2>colcon build --symlink-install
Starting >>> cv_bridge
Starting >>> image_geometry
Finished <<< image_geometry [4.98s]
Failed   <<< cv_bridge [6.61s, exited with code 1]

Summary: 1 package finished [6.91s]
  1 package failed: cv_bridge

ijnek, 2024-11-17:
Please use the ``--event-handlers console_direct+`` option to obtain more information.

ie.
```
colcon build --symlink-install --event-handlers console_direct+
```

Closing due to inactivity. Please comment if this needs to be reopened

closed 2024-11-17 by ijnek