ros-perception/vision_opencv #440: build cv_bridge failed!

huytd-pi, 2022-06-21:
I built cv_bridge from source for use with python3 following the instructions https://cyaninfinite.com/ros-cv-bridge-with-python-3/ .
but to catkin build cv_bridge error like this, how to fix it? thanks!
```
Errors     << cv_bridge:make /home/huytd/ros_catkin_ws/logs/cv_bridge/build.make.001.log
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libpython3.6m.so', needed by '/home/huytd/ros_catkin_ws/devel/.private/cv_bridge/lib/python3/dist-packages/cv_bridge/boost/cv_bridge_boost.so'.  Stop.
make[1]: *** [src/CMakeFiles/cv_bridge_boost.dir/all] Error 2
make: *** [all] Error 2
cd /home/huytd/ros_catkin_ws/build/cv_bridge; catkin build --get-env cv_bridge | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -

...............................................................................
Failed     << cv_bridge:make           [ Exited with code 2 ]                  
Failed    <<< cv_bridge                [ 0.3 seconds ]                         
[build] Summary: 0 of 1 packages succeeded.                                    
[build]   Ignored:   3 packages were skipped or are skiplisted.                
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 0.3 seconds total. 
```

ijnek, 2022-09-12:
Hi @huytd-pi, I'm the maintainer of the repo. Did you manage to solve this? If you're still looking for a solution, would you be able to provide more details about your ROS distribution and platform?

adityachechani, 2022-11-22:
I'm having the same issue. I'm using ROS Melodic on Ubuntu 18.04.

closed 2022-11-25 by huytd-pi