ros-perception/vision_opencv #443: [Enhancement] ~2x faster method to convert numpy image array to builtin array

Interpause, 2022-07-04:
https://github.com/ros-perception/vision_opencv/blob/0b69de38953f21090ac62bc90eb700e8bc98a023/cv_bridge/python/cv_bridge/core.py#L281

Looking at the above, this copies the image twice, once when converting from `np.array` to `bytes`, and another when reading into `img_msg.data` (which is an `array.array`).

Referring to <https://numpy.org/doc/stable/reference/generated/numpy.ndarray.data.html>, `np.array.data` exposes the `memoryview` of the array. By reading directly from it, it is possible to get a ~2x speed up, especially for large images (below tested using a 8k resolution image).

```py
>>> import numpy as np
>>> from array import array
>>> from timeit import timeit
>>> img = np.zeros((4320, 7680, 3), dtype=np.uint8)
>>> timeit(lambda: array('B', []).frombytes(img.tobytes()), number=200)
11.108921873000327
>>> timeit(lambda: array('B', []).frombytes(img.data), number=200)
5.20463201199982
```

I will write a pull request when I have time.

daggarwa, 2023-02-28:
@Interpause Did you happen to create an MR for this optimization? By the way I found a case where this is not working for me. If the img.data has datatype of `nd.float32` then this optimization is breaking: 

```shell
>>> import numpy as np
>>> from array import array
>>> from timeit import timeit
>>> img = np.zeros((4320, 7680, 3), dtype=np.uint8)
>>> timeit(lambda: array('B', []).frombytes(img.tobytes()), number=200)
12.889657152991276
>>> timeit(lambda: array('B', []).frombytes(img.data), number=200)
5.910853378998581
>>> img = np.zeros((4320, 7680, 3), dtype=np.float32)
>>> timeit(lambda: array('B', []).frombytes(img.data), number=200)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.8/timeit.py", line 233, in timeit
    return Timer(stmt, setup, timer, globals).timeit(number)
  File "/usr/lib/python3.8/timeit.py", line 177, in timeit
    timing = self.inner(it, self.timer)
  File "<timeit-src>", line 6, in inner
  File "<stdin>", line 1, in <lambda>
TypeError: a bytes-like object is required

```
Would you have any suggestions how to workaround that?

Interpause, 2023-03-05:
The "dtype" of a `memoryview` can be read from `memoryview.format` (https://docs.python.org/3/library/stdtypes.html#memoryview.format). So for a numpy array, it would be `img.data.format`. The formats can be referenced here: https://docs.python.org/3/library/struct.html#format-characters.

There is also a `memoryview.cast` thats can cast the `memoryview` without copy (https://docs.python.org/3/library/stdtypes.html#memoryview.cast). However, in such a case, extra precaution is needed to cast it back to the correct type when receiving.

That said, cv2 images are always supposed to be uint8 numpy arrays, so it should not be a problem?