ros-perception/vision_opencv #453: Change ros2 branch name to rolling, and updated default branch

ijnek, 2022-09-04:
In the ROS 2 Core libraries, the default branch was modified: [Change default branch name to ‘rolling’ in ROS 2 core](https://discourse.ros.org/t/change-default-branch-name-to-rolling-in-ros-2-core/26009)

I think we should do the same here, and:
- [ ] Create a ``rolling`` branch.
- [ ] Add the GitHub action to automatically sync between the new ``rolling`` branch and the old ros2 branch.
- [ ] Update the default branch from ``noetic`` to ``rolling``.
- [ ] Retarget existing PRs that target the ``ros2`` branch to target the ``rolling`` branch.

ijnek, 2022-09-12:
I don't have permission to rename branches or to change the default branch. Would you be able to either give me additional rights to do this, or do this for me? @clalancette

clalancette, 2022-09-12:
I just upgraded your rights to "Maintain", which I think should be enough to do that.

ijnek, 2022-09-12:
Thanks @clalancette

ijnek, 2022-09-13:
I've renamed ``ros2`` branch to ``rolling``

ijnek, 2022-09-13:
> People with admin permissions to a repository can change the default branch for the repository.

(https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-branches-in-your-repository/changing-the-default-branch)

It seems like I need admin access. Alternatively, it should be a one-off thing so you could just change the default branch and that should be fine.

clalancette, 2022-09-13:
> @clalancette It seems like I need admin access :(. Alternatively, it should be a one-off thing so you could just change the default branch and that should be fine.

So just to be totally clear; this repository supports both ROS 1 (via  the `noetic` branches) and ROS 2 (now via the `rolling` branch).  Currently, the `noetic` branch is the default one.  I'll change it to `rolling` if that is what you want, but I wanted to make you aware that this is more than just a branch rename.

clalancette, 2022-09-13:
It's done now.

Keep in mind that to make this totally work correctly, you'll also have to update https://github.com/ros/rosdistro/blob/master/rolling/distribution.yaml to point to the correct branch, as well as https://github.com/ros2-gbp/vision_opencv-release/blob/master/tracks.yaml#L238

ijnek, 2022-09-13:
Thanks! rosdistro has been updated in https://github.com/ros/rosdistro/pull/34463, and I've changed tracks.yaml in https://github.com/ros2-gbp/vision_opencv-release/commit/a9c882edc18e28ce8a86980253c519b8864e49a8

closed 2022-09-13 by ijnek