ros-perception/vision_opencv #485: ``foxy`` and ``galactic`` branch contains breaking changes since last release

ijnek, 2022-09-17:
When I branched off ``foxy``and ``galactic`` branches, I branched off from the ``ros2`` branch, because that was the dev branch listed in rosdistro for the distributions. However, the ``ros2`` branch contained a significant number of breaking changes, and the ``foxy``and ``galactic`` branches shouldn't have been branched off ``devel``.

Should try and revert all the breaking changes in the two branches to get the ``foxy``and ``galactic`` branches working and make a release.

ijnek, 2022-09-17:
Changes that need to be reverted:

- [ ] #352 
- [ ] #356 
- [ ] #370 
- [ ] #392 
- [ ] #397 
- [ ] #396 
- [ ] #399 
- [ ] #357 
- [ ] #400
- [ ] #388 
- [ ] #419 
- [ ] #421
- [ ] #422
- [ ] #358
- [ ] #359
- [ ] #378
- [ ] #417
- [ ] #412
- [ ] #326
- [ ] #420
- [ ] #423
- [ ] #418
- [ ] #427
- [ ] #439
- [ ] #252 
- [ ] #444 
- [ ] #451 

Changes that don't have to be reverted:

Ace314159, 2022-09-18:
Is there a reason to revert #421 and #422 as well? After the revert, builds do not work when using a Boost 1.67 or newer, as explained in https://github.com/ros-perception/vision_opencv/pull/390#issuecomment-839943369
> from [this document](https://cmake.org/cmake/help/latest/module/FindBoost.html):
> 
> > Note that Boost Python components require a Python version suffix (Boost 1.67 and later), e.g. python36 or python27 for the versions built against Python 3.6 and 2.7, respectively. This also applies to additional components using Python including mpi_python and numpy. Earlier Boost releases may use distribution-specific suffixes such as 2, 3 or 2.7. These may also be used as suffixes, but note that they are not portable.

ijnek, 2022-09-18:
Hi @Ace314159, sorry for the inconvenience the reversion has caused and thanks for reporting it.

Here's the original reasons:
#419 was reverted because that was the PR that was breaking API in foxy and galactic.
#421 was reverted because it depended a lot on changes from #419
#422 was reverted because it was fixing mistakes in #421.

However, now looking at #390, this is suggesting to me that the foxy/galactic build on windows might fail after the reversions I've made.

Could you describe the platform / distribution that's failing? And also, which branch you're building from source, (or whether you are using a binary).

Ace314159, 2022-09-18:
Thanks for the prompt response! I am building from source using the foxy branch on Windows.

ijnek, 2022-09-18:
Great, thanks for the information. Could you try out the branch: ``foxy-with-boost-python-fix`` I just put up?
It has changes from #390 merged in which were easier to include then what has happened in #421.

Ace314159, 2022-09-18:
It works! Thanks for the quick turnaround!

ijnek, 2022-09-20:
I'm going to re-open this issue until a release has been made. I'm currently trying to figure out how to get a pre-release test going before I make another release.

ijnek, 2022-10-12:
Releases were made.

However, I've been reported that the backport to galactic broke Galactic builds on RHEL: https://build.ros2.org/job/Gbin_rhel_el864__cv_bridge__rhel_8_x86_64__binary/

I've reverted the release to an older version until we resolve this. Let's track this issue here.

ijnek, 2022-10-12:
@Ace314159 could you review #495 please?

Ace314159, 2022-10-12:
It looks good to me! Just wasn't sure about the `COMPONENTS` thing I mentioned.

ijnek, 2022-10-15:
Closing, as should be fixed with #495 :crossed_fingers:

closed 2022-10-15 by ijnek