ros-perception/vision_opencv #499: ImportError: libopencv_core.so.4.2: cannot open shared object file no such file or directory

mazik1496, 2022-11-01:
`[ERROR] [1667310613.036370, 1416.343000]: bad callback: <function camera_right_callback at 0x7f477f549700>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/user/vrx_ws/src/vrx/vrx_gazebo/nodes/cam_test.py", line 55, in camera_right_callback
    cv_image2 = bridge_object.imgmsg_to_cv2(data,"bgr8")
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/core.py", line 163, in imgmsg_to_cv2
    dtype, n_channels = self.encoding_to_dtype_with_channels(img_msg.encoding)
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/core.py", line 99, in encoding_to_dtype_with_channels
    return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding))
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/core.py", line 91, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: libopencv_core.so.4.2: cannot open shared object file no such file or directory`

I am using Ubuntu 20.04 noetic!
Now, I tried importing the camera in the gazebo virtual environment that I made myself in the past with opencv, and I saw that ImportError: libopencv_core.so.4.2: was displayed.
At first, to use yolo github, I kept installing and deleting other versions of opencv.

Now, I checked libopencv_core in the admin folder, and there are files related to libopencv_core 4.4. Is there any way to open it with 4.4 instead of lipopencv_core 4.2…???

meghbhalerao, 2023-05-30:
did you figure this out? I am getting a similar error - `ImportError: libopencv_imgproc.so.407: cannot open shared object file: No such file or directory`

ngifary, 2023-06-01:
As a workaround you can create a symlink for each shared library file from the opencv version that you have libopencv*.so.4.4 with the corresponding expected file name libopencv*.so.4.2

ijnek, 2024-11-16:
Closing this issue due to inactivity. Please comment if it needs to be reopened.

If you're using Ubuntu 20.04, please install OpenCV 4.2.

"Did you figure this out? I’m getting a similar error: ImportError: libopencv_imgproc.so.407: cannot open shared object file: No such file or directory"

The error with libopencv_imgproc.so.407 seems to be an unrelated issue. I'm not sure where that version is coming from.

closed 2024-11-16 by ijnek