ros-perception/vision_opencv #509: ImportError: /lib/x86_64-linux-gnu/libp11-kit.so.0: undefined symbol: ffi_type_pointer, version LIBFFI_BASE_7.0

TKroos18, 2023-02-16:
When I use some function in cv_bridge, I receive an importerror:

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 76, in callback
    self.signalMessage(msg)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 240, in add
    self.signalMessage(*msgs)
  File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "stretch_ws/src/grasp_logic/scripts/main.py", line 22, in callback
    color_image = bridge.imgmsg_to_cv2(data1, 'bgr8')
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/core.py", line 163, in imgmsg_to_cv2
    dtype, n_channels = self.encoding_to_dtype_with_channels(img_msg.encoding)
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/core.py", line 99, in encoding_to_dtype_with_channels
    return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding))
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/core.py", line 91, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: /lib/x86_64-linux-gnu/libp11-kit.so.0: undefined symbol: ffi_type_pointer, version LIBFFI_BASE_7.0

The version of os is Ubuntu 20.04. The version of python is 3.8.16. I create virtual environment with anconda.
So what happened and is there a way to use CV-Bridge within conda virtual environment with ROS-Noetic?

bqm1111, 2024-01-29:
@TKroos18 Did you figure out how to fix it?

TKroos18, 2024-02-05:
> @TKroos18 Did you figure out how to fix it?

I downgrade my python to 3.8.10.

MyunghyunKim, 2024-07-17:
In my case relink the library
my python ver = 3.8.19

$ ln -sf /usr/lib/x86_64-linux-gnu/libffi.so.7 ~/anaconda3/envs/**[your env name here]**/lib/libffi.so.7

ijnek, 2024-11-16:
I am going to close out this issue given that:
* pythonversions > 3.8.10 are not officially supported for noetic
* This is more a problem with how libp11-kit is being loaded from the base OS, while libffi (from version 8) is being loaded from within the conda environment.
Please either downgrade to 3.8.10, or use the workaround suggested by @MyunghyunKim to create a symbolic link (ln -sf) for the libffi.so.7 library from the system directory (/usr/lib/x86_64-linux-gnu/) to the specified libffi.so.7 file in the lib directory of your conda environment.

closed 2024-11-16 by ijnek

UUwei-zuo, 2025-01-22:
I solved this by:
https://medium.com/@gz285639/fixing-libp11-kit-so-0-undefined-symbol-ffi-type-pointer-in-ros-and-carla-integration-0aef01d7c6ff