ros-perception/vision_opencv #524: image_geometry/cameramodels.py uses deprecated numpy.matrix subclass

ScottMonaghan, 2024-03-23:
I'd be happy to try to tackle solving and creating a PR for this issue, but I have some questions before I proceed.

### Background
This issue arises from `PendingDeprecationWarning` reported in [ros-perception/image_pipeline issue #572](https://github.com/ros-perception/image_pipeline/issues/572).

### Issue Summary
image_geometry/cameramodels.py uses deprecated `numpy.matrix` subclass causing runtime `PendingDeprecationWarning`. 

The use of `numpy.matrix` should be deprecated and/or replaced.

### Details

`PinholeCamera`'s exposed members--`K`, `D`,`R`,`P`,`full_K`, and `full_P` are initialized as `numpy.matrix` objects:

https://github.com/ros-perception/vision_opencv/blob/8b42d66f4946f4204387b30aabc8c2921e5d9f47/image_geometry/image_geometry/cameramodels.py#L6-L9

...

https://github.com/ros-perception/vision_opencv/blob/8b42d66f4946f4204387b30aabc8c2921e5d9f47/image_geometry/image_geometry/cameramodels.py#L32-L47

Without much work we could substitute 'ndarray' for 'numpy.matrix', and also update any unsupported matrix math operators ([details](https://stackoverflow.com/a/53254739)) in the `PinholeCamera` class.

### Should we remove or deprecate?

I'm new to the ROS 2 community so I'm not sure what the best practice is here.

The concern I have is that all of the class members in question are exposed, and therefore could be in use by any other package--expecting `numpy.matrix` objects, and as ROS 2 (while young) has had official releases, I'd consider these exposed members to be part of the ROS 2 api.

In a perfect world, I'd recommend that we:
 - deprecate `K`, `D`,`R`,`P`,`full_K`, and `full_P`
 - create getters that throw a `PendingDeprecationWarning` when they are called
 - create replacement members that return as  `numpy.ndarray` (with more intuitive names)
   - `ndarray:PinholeCamera.K_camera_matrix`
   - `ndarray:PinholeCamera.R_rectification_matrix`
   - `ndarray:PinholeCamera.P_projection_matrix`
   - `ndarray:PinholeCamera.D_distortion_parameters`
   - `ndarray:PinholeCamera.full_K_camera_matrix`
   - `ndarray:PinholeCamera.full_P_projection_matrix`

As noted before, I'm new to the ROS community and even if the above isn't overkill, I'm sure veteran community members would prefer some of my suggestions be done differently. I'm definitely open to feedback.

@mikeferguson, since this is the root cause of [ros-perception/image_pipeline issue #572](https://github.com/ros-perception/image_pipeline/issues/572), which you marked as 'good first issue'. Likewise, I apologize in advance if direct @ tagging is frowned upon.

ScottMonaghan, 2024-03-24:
Ah! After some reading, since the image_geometry class doesn't have a declared API, we've got some leeway in the rules for these changes.

From the [ROS 2 Developer Guide](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Developer-Guide.html): 
> Any elements of the public API should be clearly stated in the package’s documentation. If something you are using is not explicitly listed as part of the public API in the package’s documentation, then you cannot depend on it not changing between minor or patch versions.

That said I do think it makes sense to deprecate these 'PinholeCameraModel' methods as they'll now have a new return type, and they're clearly meant to be Public:
- deprecate & replace `‎intrinsicMatrix(self)->numpy.matrix` with `‎intrinsic_matrix(self)->numpy.ndarray` 
- deprecate & replace `projectionMatrix(self)->numpy.matrix` with `‎projection_matrix(self)->numpy.ndarray` 
- deprecate & replace `‎distortionCoeffs(self)->numpy.matrix` with `‎distortion_coeffs(self)->numpy.ndarray` 
- deprecate & replace `rotationMatrix(self)->numpy.matrix` with `rotation_matrix(self)->numpy.ndarray` 
- deprecate & replace `fullIntrinsicMatrix(self)->numpy.matrix` with `full_intrinsic_matrix(self)->numpy.ndarray` 
- deprecate & replace `fullProjectionMatrix(self)->numpy.matrix` with `full_projection_matrix(self)->numpy.ndarray` 

Then for consistency & style sake, I'd like to update the rest of the methods accordingly: deprecating camelCase and replacing them with snake_case to fit the style guidelines, but if you all think that's too much, I'll hold off on that.

ijnek, 2024-04-05:
Hi @ScottMonaghan, sorry for the late reply and thanks for the nicely written issue. 

Everything in general looks good, strategies for preventing API breakage are good too. Please use `DeprecationWarning` instead of `PendingDeprecationWarning`. Let's aim to deprecate in ROS 2 Jazzy, and completely remove in ROS 2 K-turtle.

To maintain functionality of deprecated API, I suggest updating underlying numpy data structure, and do a conversion to numpy.matrix upon request. The new methods can have the same name, but using snake case instead of camel case. Please add a note in the deprecation warning saying that the returned numpy data type of the new method is different.

ScottMonaghan, 2024-04-05:
@ijnek, no problem with the delay. This past week was an object lesson in why we all need to treat our volunteer FOSS maintainers with gratitude.

Regarding time, I'm going to target getting a PR submitted by end of April; this is my first ROS contribution, and while I've installed ROS in hello-world mode, I expect I might have some trial and error before I can get my fork to build and play nicely.

If you have any newbie first-time time tips, I'm all ears. Otherwise, I'll dive back into the documentation and get to it.

ScottMonaghan, 2024-04-20:
@ijnek, with the merging of PR #527, this issue should be able to be closed.

closed 2024-04-20 by ijnek