ros-perception/vision_opencv #530: Rolling Ubuntu Jammy image_geometry build disabled since 3-7-2024

ScottMonaghan, 2024-04-21:
@ijnek, I'm not sure if this is intentional, but I just wanted to bring it up in case you aren't aware.
As of writing, the Jammy image_geometry build for rolling is currently disabled since multiple failures on March 7, 2024.

I did see the noble build is succeeding.

Should the jammy build be working or is rolling moving to noble (in which case I will update my dev environment accordingly).

https://build.ros2.org/job/Rsrc_uJ__image_geometry__ubuntu_jammy__source/

[image]

ScottMonaghan, 2024-04-21:
Following up on this I see that Rolling is currently transitioning into noble.

https://discourse.ros.org/t/preparing-ros-2-rolling-for-the-transition-to-ubuntu-24-04/35673/17

@ijnek, is your plan to support noble moving forward and cease support for Jammy (with Rolling releases)?

ijnek, 2024-04-29:
@ScottMonaghan thanks for raising this issue. I was unaware that the builds were failing, but I see the [source builds for rolling on noble](https://build.ros2.org/job/Jsrc_uN__image_geometry__ubuntu_noble__source/) are passing, so it seems that this issue is resolved.

You're right, rolling is now meant to be on ubuntu noble 24.04 and support for Jammy is dropped. No future releases will/ can be made on rolling for jammy.

ScottMonaghan, 2024-04-29:
I just upgraded the other night to the noble nightly build.

Anything else you'd like me to take a look at in the image_geometry or overall repo?

I'm much stronger with Python than C++, but I do work with both professionally--for better or worse:)

ScottMonaghan, 2024-04-29:
Thanks @ijnek!

I just upgraded the other night to the noble nightly build.

Anything else you'd like me to take a look at in the image_geometry or overall repo?

I'm much stronger with Python than C++, but I do work with both professionally--for better or worse:)