ros-perception/vision_opencv #537: [FYI] A workaound for cv_bridge_boost's ImportError on Windows

sgr-aoba, 2024-10-26:
This is a workaround for cv_bridge_boost's ImportError with ROS2 on Windows (humble).
I hope this helps anyone encountering the same error.

# Environment

| Component | Version |     |
| --------- | ------- | --- |
| Operating System | Windows 10 22H2 |
| ROS2 Release | Humble | Binary installation by following [the guide](https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html). |
| vision_opencv | [Humble branch](https://github.com/ros-perception/vision_opencv/tree/humble) | Installed as a package (by git submodule).  And set the environment variable `OpenCV_DIR`. |
| boost | 1.74.0 (msvc-14.2-64) | Binary installation from [sourceforge](https://sourceforge.net/projects/boost/files/boost-binaries/1.74.0/). And set the environment variable `BOOST_ROOT`. |

# Steps to reproduce issue

- Create a python node using CvBridge
- Build (colcon build)
- Run

# Expected behavior

Nodes using CvBridge work as expected.

# Actual behavior

Nodes using CvBridge will exit with the following error.

```
ImportError: DLL load failed while importing cv_bridge_boost: The specified module could not be found.
```

# Workaround

It need to add DLL paths [explicitly](https://docs.python.org/3.8/library/os.html#os.add_dll_directory).
Add the following snippet before importing CvBridge.

```python
import platform
if platform.system() == 'Windows':
    import os
    from pathlib import Path
    from ament_index_python.packages import get_package_prefix
    dll_dir: Callable[[Path], Path] = lambda p: next(p.rglob('*.dll')).parent
    for d in [
        Path(os.environ.get('BOOST_ROOT')),
        Path(os.environ.get('OPENCV_DIR')),
        Path(get_package_prefix('cv_bridge')),
    ]:
        os.add_dll_directory(dll_dir(d))

from cv_bridge import CvBridge, CvBridgeError
```

sgr-aoba, 2024-10-31:
Related issues: #247, ms-iot/ROSOnWindows#371

sgr-aoba, 2025-01-23:
Have you built cv_bridge?
ROS for Windows does not include cv_bridge.

sgr-aoba, 2025-01-24:
It seems that your problem is a issue with building of cv_bridge.
The introduction above assumes that cv_bridge is available as a package.

How about trying git clone ros-perception to `<YOUR_ROS2_PROJECT>/src`, and run `colcon build`.

ijnek, 2025-06-14:
Thanks for posting this workaround for others! I'm going to close it as #247 tracks the actual issue.

closed 2025-06-14 by ijnek