ros-perception/vision_opencv #545: ros-humble，opencv4.5.4，orbslam3_ros2，undefined reference to symbol '_ZN2cv23initUndistortRectifyMapERKNS_11_InputArrayES2_S2_S2_NS_5Size_IiEEiRKNS_12_OutputArrayES7_'

clhchan, 2025-03-02:
**When I was compiling the orb_slamros package, an error occurred. The c++ program is probably a process of converting topic data into images using cv_bridge. The error encountered the problem that cv_bridge could not find the definition symbol.

My environment is ubuntu22.04, opencv4.5.4, cv_bridge. I tried to install it through apt install ros-humble-cv_bridge, and also tried to compile the source code and specify my current opencv4.5.4 in Cmakelist.txt for compilation, but in the end, compiling my c++ program package again will still report an error. I searched online and it was probably because the opencv version and cv_bridge are incompatible?

After I installed ros-humble, this opencv4.5.4 version is the system's own version. What version of opencv does ros-humblecv_bridge need?**
I even tried opencv4.2.0, that version also didn't work.

[image]

 59 |         KannalaBrandt8(KannalaBrandt8* pKannala) : GeometricCamera(pKannala->mvParameters), precision(pKannala->precision), mvLappingArea(2,0) ,tvr(nullptr) {
      |         ^~~~~~~~~~~~~~
/usr/bin/ld: warning: libopencv_imgcodecs.so.4.5, needed by /opt/ros/humble/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.4.5d
/usr/bin/ld: warning: libopencv_imgcodecs.so.4.5, needed by /opt/ros/humble/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.4.5d
/usr/bin/ld: warning: libopencv_imgproc.so.4.5, needed by /opt/ros/humble/lib/libcv_bridge.so, may conflict with libopencv_imgproc.so.4.5d
/usr/bin/ld: warning: libopencv_core.so.4.5, needed by /opt/ros/humble/lib/libcv_bridge.so, may conflict with libopencv_core.so.4.5d
/usr/bin/ld: CMakeFiles/stereo.dir/src/stereo/stereo-slam-node.cpp.o: undefined reference to symbol '_ZN2cv23initUndistortRectifyMapERKNS_11_InputArrayES2_S2_S2_NS_5Size_IiEEiRKNS_12_OutputArrayES7_'
/usr/bin/ld: /lib/x86_64-linux-gnu/libopencv_calib3d.so.4.5d: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/stereo.dir/build.make:217: stereo] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:195: CMakeFiles/stereo.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: warning: libopencv_imgcodecs.so.4.5, needed by /opt/ros/humble/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.4.5d
/usr/bin/ld: warning: libopencv_imgproc.so.4.5, needed by /opt/ros/humble/lib/libcv_bridge.so, may conflict with libopencv_imgproc.so.4.5d
/usr/bin/ld: warning: libopencv_core.so.4.5, needed by /opt/ros/humble/lib/libcv_bridge.so, may conflict with libopencv_core.so.4.5d
/usr/bin/ld: warning: libopencv_imgcodecs.so.4.5, needed by /opt/ros/humble/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.4.5d
/usr/bin/ld: CMakeFiles/stereo-inertial.dir/src/stereo-inertial/stereo-inertial-node.cpp.o: undefined reference to symbol '_ZN2cv23initUndistortRectifyMapERKNS_11_InputArrayES2_S2_S2_NS_5Size_IiEEiRKNS_12_OutputArrayES7_'
/usr/bin/ld: /lib/x86_64-linux-gnu/libopencv_calib3d.so.4.5d: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/stereo-inertial.dir/build.make:216: stereo-inertial] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:221: CMakeFiles/stereo-inertial.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< orbslam3 [1min 9s, exited with code 2]

Summary: 0 packages finished [1min 10s]
  1 package failed: orbslam3
  1 package had stderr output: orbslam3

LeoAquinas, 2025-04-23:
[https://github.com/alsora/ros2-ORB_SLAM2/issues/8](url)

ijnek, 2025-06-14:
Closing, as I suspect this is not a problem with this repository, but related to the ros2-ORB_SLAM2 repository as @LeoAquinas posted.