ros-perception/vision_opencv #546: ROS2 cross_compile mistake :   Could NOT find Python3 (missing: Python3_NumPy_INCLUDE_DIRS NumPy) (found   version "3.10.12")

Dongzh0915, 2025-03-25:
[image]
env: ubuntu22.04 、cmake 3.22.1、python 3.10.12、NumPy 1.26.01

Dongzh0915, 2025-05-16:
> I have the same problem, but I'm trying to build on gentoo linux.
> 
> Anyway, `find_package(Python3 ... Numpy)` doesn't seem to support cross-compile because this method internally executing python (like import numpy; ...), so not compatible with cross-compile.

Yes, so in the end I used the arm64 architecture docker image to complete the compilation, directly skipping the cross-compilation process. ROS2's support for cross-compilation is indeed not very friendly.

XQSwx, 2025-05-19:
I also encountered this problem in the cross-compilation environment. Are there any solutions?

XQSwx, 2025-05-19:
This is okay, but it doesn't fundamentally solve this problem. There are other packages that can find the numpy module. Why can't it find it

seungin-cha, 2025-05-19:
> This is okay, but it doesn't fundamentally solve this problem. There are other packages that can find the numpy module. Why can't it find it

Actually, I didn't analyze deeply but the cause seems to be https://github.com/ros-perception/vision_opencv/issues/546#issuecomment-2883124057.

XQSwx, 2025-05-19:
I have blocked the python part of this package now