ros-perception/vision_opencv #553: No jazzy branch?

tranzmatt, 2025-08-12:
I was trying to build a jazzy version of ros2 on my RPI5 running bookworm with an Oak-D Lite.  The core builds fine, but thinks balk here because of vision_opencv, but there's no jazzy branch or tag to checkout.   Is ROS2 Jazzy just not supported?  Is there a workaround otherwise?

InvincibleRMC, 2025-12-06:
Looking at https://github.com/ros2-gbp/vision_opencv-release/blob/master/tracks.yaml since vision_opencv has a release for jazzy and kilted. The buildfarm is building from rolling branch. What was the failure?

InvincibleRMC, 2025-12-17:
@ijnek Any chance we could add `jazzy` and `kilted` branches? I'm working on some features see #557, #558, #560 and would to not have to worry about breaking backwards compatibility.